moveit/moveit2 #1072: Mixed up implementations createStaticTfSubscription <-> createDynamicTfSubscription

ruelj2, 2022-02-10:
Function naming is mixed up between [line 97](https://github.com/ros-planning/moveit2/blob/77e43f339cb6ee06679a424e21c72301797920d4/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor_middleware_handle.cpp#L97) and [line 91](https://github.com/ros-planning/moveit2/blob/77e43f339cb6ee06679a424e21c72301797920d4/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor_middleware_handle.cpp#L91) in `CurrentStateMonitorMiddlewareHandle`. Should be 
```
void CurrentStateMonitorMiddlewareHandle::createDynamicTfSubscription(TfCallback callback)
{
  static_transform_subscriber_ =
      node_->create_subscription<tf2_msgs::msg::TFMessage>("/tf", tf2_ros::DynamicListenerQoS(), callback);
}

void CurrentStateMonitorMiddlewareHandle::createStaticTfSubscription(TfCallback callback)
{
  transform_subscriber_ =
      node_->create_subscription<tf2_msgs::msg::TFMessage>("/tf_static", tf2_ros::StaticListenerQoS(), callback);
}
```

vatanaksoytezer, 2022-02-10:
Hey @ruelj2 thanks for reporting. That looks wrong, yeah. Do you mind opening up a PR, so we can just merge?

ruelj2, 2022-02-21:
Closed since #1073 has been merged to main.

closed 2022-02-21 by ruelj2