moveit/moveit2 #1075: OMPL "request adapters" parameters aren't read properly

AndyZe, 2022-02-14:
### Description

Modifying the "request adapter" in a MoveGroup launch file does not have any effect.

### Your environment
* ROS Distro: ROS2 Rolling
* Source build, main branch

### Steps to reproduce
Edit this line in panda_moveit_config/demo.launch.py:

`"request_adapters": """default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/ResolveConstraintFrames default_planner_request_adapters/FixWorkspaceBounds default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints""",`

Change it to something invalid like this:

`"request_adapters": """test""",`

Then run the launch file:

`ros2 launch moveit_resources_panda_moveit_config demo.launch.py`

You should see some terminal output indicating it is still using the default request adapters:

```
[move_group-4] [INFO] [1644848949.048494917] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Parameterization'
[move_group-4] [INFO] [1644848949.048507989] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[move_group-4] [INFO] [1644848949.048512591] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[move_group-4] [INFO] [1644848949.048516398] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[move_group-4] [INFO] [1644848949.048519733] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[move_group-4] [INFO] [1644848949.048523060] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
```

AndyZe, 2022-02-14:
This isn't caused by any recent commit. I went back to version 2.4.0 and it has the same issue.

In October 31 2021 it was working properly. (I know so because of [this PR](https://github.com/ros-planning/moveit2/pull/712)).

henningkayser, 2022-02-14:
I'm not sure `move_group` is the right namespace [here](https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/launch/demo.launch.py#L69). I would imagine that the planning pipeline is simply initialized to ompl by default and the provided parameter does nothing. Please try using the mapped pipeline names to make sure that the parameters are loaded correctly, or better use moveit_configs_utils like in the tutorials https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/launch/demo.launch.py#L69 and configure the request adapters in the yaml file https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/ompl_planning.yaml#L2

closed 2023-01-26 by henningkayser