moveit/moveit2 #1082: Refactoring and Testing Calculations

tylerjw, 2022-02-18:
### Need

As a user of MoveIt
I want routines for calculations to be separated from side-affect-causing code so it can be re-used and tested easily.

### Acceptance

4 PRs landed that:
* refactor out a calculation from a section of code that has side effects to a pure function
* add a test of that pure function

### Notes

I'm defining a `Calculation` as a block of code that always produces the same output given the same inputs and is not dependent on the number of times it is run.  This means that Calculations are pure sections of code and do not depend on or cause side effects.

tylerjw, 2022-02-21:
@mamoll suggested extracting Signed Distance Fields from CHOMP.  Please edit this comment to clarify.

litmus-zhang, 2022-06-14:
Hi @tylerjw  how can i also contribute to this?

Mech-Mon, 2022-09-30:
Hello @tylerjw , I am interested in helping out. Noob to contributing.

TaMaN2031A, 2023-03-24:
Good morning,
If I could ask, where can I find this "calculation" part? or do you mean that this problem is widespread in the whole code?
Thank you.

Abhishek-k-kori, 2023-04-03:
I m intrested to help you through GSOC ,

Henry-YiW, 2023-04-04:
Hi, I'm very interested in this project through GSOC.

tylerjw, 2023-08-23:
I'm closing this as it was an odd issue that is not specific enough.

closed 2023-08-23 by tylerjw