moveit/moveit2 #1083: Interfaces Use Strong Types for Prameters

tylerjw, 2022-02-18:
### Need

As a developer using MoveIt;
I want MoveIt to have expressive interfaces that can only be used correctly.

### Acceptance

A PR is merged into moveit_core that introduces the concept of strong types for interfaces.

### Notes

https://www.fluentcpp.com/2016/12/08/strong-types-for-strong-interfaces/

gavanderhoorn, 2022-02-19:
Somewhat off-topic, but: the linked page demonstrates a nice idea, but I'm wondering whether he isn't confusing type with role.

A *width* is not a type in my opinion. It's a regular `double` which is used in a certain context to represent a certain quantity with certain semantics. It's still a number, but it plays a different *role* as a `double` used to represent a height. Is that second `double` of a different class of numbers?

The fact we need to redeclare/reimplement all the regular mathematical operators and cannot easily use a `Width` as a regular number any more (which it basically is) seems to indicate something is not right with this approach. Enabling implicit conversion (as discussed in the comments) is one way to deal with this, but that seems to negate the main point of using the strong types (ie: it's semantically and syntactically clear what type is being used, and the compiler can check it for us).

Using user-defined literals to assign units to everything is something I see much more value in, and doesn't try to use typing to capture roles.

tylerjw, 2022-02-19:
> Perhaps the concept system would be better suited to capture these kind of things?

Concepts are amazing, but it will probably be years before we can use them as we'll need ROS to switch to C++20 before we can.

I'll try to find an example of where I think this could be useful in MoveIt to help explain this idea.  On a client project I was just on I used it for file descriptors and it made an interface much clearer than if it was just an int.

gavanderhoorn, 2022-02-19:
Perhaps the example in the linked article is just wrong.

A file descriptor is not an `int`. It's a wholly different concept, which just happens to be represented by an `int`. But the `int` is really an index into a table/list of entities managed by the OS.

A *width* is not something which is conceptually similar.

A width stays a number, is reasoned about as a number, can be used in mathematical expressions as a number and that all makes sense.

No one tries to add two file descriptors together, as that would not make sense.

But it can make sense to want to do: `width + height + width + height` (for instance to calculate a circumference). If `Width` were an actual (strong) type, I'd have to either rely on implicit conversion, or overload the `+` to allow adding `Width` to `Height`.

tylerjw, 2022-02-19:
I agree that width and height are maybe bad examples.  For whatever reason if you Google strong types it seems to be the go-to example everyone uses.  To reduce a ton of boilerplate you would need for strong types there is this library by the author of that blog: https://github.com/joboccara/NamedType

In the case of my file descriptor it made sense to me because it is a very different thing from an int, it just happens to be represented as an int in the language.  As you said it makes no sense to multiply a file descriptor or really to do any math operations on it other than comparing it and checking the sign to see if it is valid.

I'll look for examples in MoveIt where this would improve the readability of the interfaces as that is the clear benefit I see to using strong types.

tylerjw, 2022-02-19:
An example that might make sense (I'd need to try to make the change somewhere to understand if it would) would be velocity and position as strong types.  There are cases where the two intact in math operations but it is not the full set of operations that can be done on two doubles. When they do interact they have a specific type that results.

For example it makes no sense to add a position to a velocity so maybe it would make sense to build types that would allow the compiler catch that.

gavanderhoorn, 2022-02-19:
Right, so that's what I was referring to with:

> Using user-defined literals to assign units to everything is something I see much more value in, and doesn't try to use typing to capture roles.

Things like [boost.units](https://www.boost.org/doc/libs/1_78_0/doc/html/boost_units.html) and [nholthaus/units](https://github.com/nholthaus/units) support what you describe.

tylerjw, 2022-02-19:
I think I finally understand what you mean by roles.  I think we see value in the idea in the same place.  I'll go check out the tools you linked.

tylerjw, 2022-02-19:
That units library looks awesome.  I'm going to vendor it and just start using it if it isn't in rosdep already.  That is basically was I was hoping for and and so much more.

henningkayser, 2022-04-13:
@peterdavidfagan did you already spent time on this? I agree that it's something that could have an effect on the GSoC project, but I'm not sure if we already have a good understanding on what API changes we actually need. I also think it's not really a "good first issue" anymore, so I removed the label. @tylerjw do you have any more thoughts on this?

peterdavidfagan, 2022-04-13:
Hi @henningkayser, I haven't spent much time on this issue yet other than briefly looking into the libraries mentioned above. I am happy to stop working on this in the meantime given the above comment.

chiphogg, 2024-07-03:
Hi!  I recently stumbled on this issue, and had some suggestions to share.

If you do end up adding interfaces with strongly-typed units, you'll need to pick a C++ units library... which means you'll need to compare the options.  This [comparison of alternatives](https://aurora-opensource.github.io/au/main/alternatives/) dives deep into the C++ units libraries that have the most GitHub stars, plus Boost Units (which was available _before GitHub existed_).

Full disclosure of my bias: I'm the lead author for one of those libraries, [aurora-opensource/au](https://github.com/aurora-opensource/au).  The comparison page is hosted on Au's doc website.  Au wasn't publicly available when this issue was created; it was open sourced in Feb 2023.  But it does come from a background being used in production for years in robotics applications (namely, self-driving trucks), which is why I thought it might be a good fit here.