moveit/moveit2 #1089: UR10 + IKFAST problem

QNXLQ, 2022-02-23:
### Description

when i try to simulate UR10 with IKFAST in rviz, the node [move_group-4] process has died and [rviz_ubuntu_11518_7826013209425677446-5] process has died  too.

### Your environment
* ROS Distro:Melodic
* OS Version:  Ubuntu 18.04

### Steps to reproduce
Follow the http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html
Create the `ur10.urdf` and `ur10.dae` form `rosrun xacro xacro --inorder -o ur10.urdf ur10_joint_limited_robot.urdf.xacro` and `rosrun collada_urdf urdf_to_collada ur10.urdf ur10.dae`

Check the ur10.dae by `openrave ur10.dae` and `openrave-robot.py --info liks`

Generaet IK Solver `python `openrave-config --python-dir`/openravepy/_openravepy_/ikfast.py --robot="$MYROBOT_NAME".dae --iktype=transform6d --baselink=1 --eelink=10 --savefile=ur10_manipulator_ikfast_solver.cpp`

At last create a new package and create plugin by `rosrun moveit_kinematics create_ikfast_moveit_plugin.py ur10 manipulator  ur10_moveit_plugins base_link tool0 ur10_manipulator_ikfast_solver.cpp`

In this moment, the `kinematics.yaml` is be modified: `kinematics_solver: ur10_manipulator/IKFastKinematicsPlugin`
### Expected behaviour
when i run `roslaunch ur10_moveit_config demo.launch` should be run, open the rviz and use the IKFAST but not OMPL

### Actual behaviour
when i run `roslaunch ur10_moveit_config demo.launch` , existe some WARN and some node or process has died:

`process[rviz_ubuntu_12058_730513676039804151-5]: started with pid [12099]`
`[ WARN] [1645610756.144692545]: Falling back to using the move_group node's namespace (deprecated Melodic behavior).`
`[ INFO] [1645610756.152780482]: Loading robot model 'ur10'...`
`[ WARN] [1645610756.215240592]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.`
`[ INFO] [1645610756.239200731]: Using empty link_prefix.
/home/nielei/catkin_ws/devel/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/nielei/catkin_ws/devel/lib//libur10_manipulator_moveit_ikfast_plugin.so: undefined symbol: _ZN16ur10_manipulator17GetFreeParametersEv`
`[ INFO] [1645610756.296457674]: rviz version 1.13.23`
`[ INFO] [1645610756.296563893]: compiled against Qt version 5.9.5`
`[ INFO] [1645610756.296581675]: compiled against OGRE version 1.9.0 (Ghadamon)`
`[ INFO] [1645610756.304489190]: Forcing OpenGl version 0.`
`[move_group-4] process has died [pid 12093, exit code 127, cmd /home/nielei/catkin_ws/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/nielei/.ros/log/30058384-9490-11ec-914d-000c29873124/move_group-4.log].`
`log file: /home/nielei/.ros/log/30058384-9490-11ec-914d-000c29873124/move_group-4*.log`
(by the way, in this path i can't find that log file)
`[ INFO] [1645610757.964151225]: Stereo is NOT SUPPORTED`
`[ INFO] [1645610757.964351472]: OpenGL device: SVGA3D; build: RELEASE;  LLVM;`
`[ INFO] [1645610757.964464293]: OpenGl version: 3.3 (GLSL 3.3) limited to GLSL 1.4 on Mesa system.`
`[ INFO] [1645610761.372965106]: Loading robot model 'ur10'...`
`[ WARN] [1645610761.413350082]: Kinematics solver doesn't support #attempts anymore, but only a timeout.`
`Please remove the parameter '/rviz_ubuntu_12058_730513676039804151/manipulator/kinematics_solver_attempts' from your configuration.`
`[ INFO] [1645610761.438200419]: Using empty link_prefix.`
`/opt/ros/melodic/lib/rviz/rviz: symbol lookup error: /home/nielei/catkin_ws/devel/lib//libur10_manipulator_moveit_ikfast_plugin.so: undefined symbol: _ZN16ur10_manipulator17GetFreeParametersEv`
`[rviz_ubuntu_12058_730513676039804151-5] process has died [pid 12099, exit code 127, cmd /opt/ros/melodic/lib/rviz/rviz -d /home/nielei/catkin_ws/src/universal_robot/ur10_moveit_config/launch/moveit.rviz __name:=rviz_ubuntu_12058_730513676039804151 __log:=/home/nielei/.ros/log/30058384-9490-11ec-914d-000c29873124/rviz_ubuntu_12058_730513676039804151-5.log].`
(this log file is not existe in this paths, neither)

I did not change any thing manually, only the file `kinematics.yaml` changed by the code `rosrun moveit_kinematics create_ikfast_moveit_plugin.py ur10 manipulator  ur10_moveit_plugins base_link tool0 ur10_manipulator_ikfast_solver.cpp` automatically.

rbtchun3, 2022-03-07:
Hey,

I have the same error and came across this but am not sure if this is a good/viable solution.

https://answers.ros.org/question/364020/moveit-crash-undefined-symbol-in-libmoveit_kdl_kinematics_pluginso/

henningkayser, 2023-01-26:
@QNXLQ I'm weeding out stale issues. Have you found a solution to your problem? If yes, please close, otherwise an update would be helpful. Since you are running melodic, I'm transferring this to ros-planning/moveit