moveit/moveit2 #1091: Planning Scene Monitor settings should all be specified in one place

AndyZe, 2022-02-23:
This is an observation that the planning scene monitor [PSM] settings are scattered so it's hard to figure out exactly how to configure your PSM. You really have to read the header file carefully. For example, topic names are often given in a yaml file but you need to call these functions from code:

```
  planning_scene_monitor_->startSceneMonitor("/monitored_planning_scene");
  planning_scene_monitor_->startWorldGeometryMonitor("/collision_object");
  planning_scene_monitor_->startStateMonitor("/joint_states", "/attached_collision_object");
  planning_scene_monitor_->monitorDiffs(true);  // Often forgotten
  planning_scene_monitor_->stopPublishingPlanningScene();
```

I think it would be great if this configuration were all specified in one place, probably a yaml file.

It also would be good to get clarification on what happens if `startSceneMonitor` and `startStateMonitor` have both been called. Do joint angles from `joint_states` take precedence, or do the `/monitored_planning_scene` joint angles take precedence?

henningkayser, 2022-03-18:
The monitors are different things, they don't interfere with each other. I've been thinking about splitting up the PSM to some extend, but any kind of change would involve a bigger effort. We should definitely improve documentation, though.
I don't think that these topics should be configurable, actually. (I'm aware that I did just that for MoveItCpp back then). They should rather be treated as the public ROS interface of the PSM with constant names so that it's easier to reason about the different PSM instances. Instead, we should simply configure the namespace of the other PSM instance. If they would all use arbitrary topic names, it's impossible to understand what's going on.

>It also would be good to get clarification on what happens if startSceneMonitor and startStateMonitor have both been called. Do joint angles from joint_states take precedence, or do the /monitored_planning_scene joint angles take precedence?

The scene monitor only updates the joint state if [UPDATE_STATE](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L74) is enabled and if the RobotState is flagged as an update with [is_diff](https://github.com/ros-planning/moveit_msgs/blob/ros2/msg/RobotState.msg#L13). If the other scene is configured to publish joint state updates, the incoming scene updates from `/monitored_planning_scene` and the ones from the `/joint_states` topic are both being applied by the different monitors. None takes precedence, they are both applied in order of receiving. However, the time stamp must be increasing, the state will not move back in time.

v4hn, 2022-03-19:
I agree to an options struct for PSM creation. even if we keep the topic names constant (good point henning), there are some behavior decisions in the class that could be configured during construction, together with a name/namespace.

it's actually an annoying inconsistency that the CSM does not take precedence when enabled. in practice it "mostly" does, because joint states come in with a much higher frequency...