moveit/moveit2 #1094: error while loading libbackward.so

aaryanmurgunde, 2022-02-27:
### Description
When the move_group node is executed, it immidiatly dies with the following error `moveit_ros_move_group/move_group: error while loading shared libraries: libbackward.so: cannot open shared object file: No such file or directory`

[image]

This seems to be caused by the missing package ros_backwards.
**This dependency should have been installed directly by the binary itself.**

### Fix
`sudo apt install ros-galactic-backward-ros` or `sudo apt install ros-<ROS DISTRO>-backward-ros`

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04
* Binary: Package: ros-galactic-moveit
Version: 2.3.4-1focal.20220208.011441

JafarAbdi, 2022-03-08:
I'm not %100 sure, but I think it's because we have moveit_common as a `build_depend` rather than `depend`

v4hn, 2022-03-09:
On Tue, Mar 08, 2022 at 06:02:04AM -0800, Jafar Abdi wrote:
> I'm not %100 sure, but I think it's because we have moveit_common as a `build_depend` rather than `depend`

No, the problem is that libbackward.so & backward_ros are installed on the buildfarm when the package is built, so cmake picks them up.
We wanted to have this linked only when building yourself with backward_ros installed on your system instead of making it a fixed dependency.

@jspricke probably can support here.

jspricke, 2022-03-10:
>On Tue, Mar 08, 2022 at 06:02:04AM -0800, Jafar Abdi wrote:
>> I'm not %100 sure, but I think it's because we have moveit_common as a `build_depend` rather than `depend`

Correct, plus that you inject backward_ros everywhere:

https://github.com/ros-planning/moveit2/blob/1faa2b16c94271a9290a6106181f8ef856f9a4ca/moveit_common/cmake/moveit_package.cmake#L36

>No, the problem is that libbackward.so & backward_ros are installed on the buildfarm when the package is built, so cmake picks them up.
>We wanted to have this linked only when building yourself with backward_ros installed on your system instead of making it a fixed dependency.

Then you should drop the dependency in the package.xml:

https://github.com/ros-planning/moveit2/blob/4c75dcf0bf4c98e1baf56765095b69d4297aabcf/moveit_common/package.xml#L21

tylerjw, 2022-03-10:
From my perspective, I don't see a problem injecting backward_ros into our release binaries.  My intention was that it would always be there in the releases and not that it would require installing it separately.

v4hn, 2022-03-11:
Seems there was some disagreement then on the original intention. :-)
I don't care much either way. I just think it is much more useful with a `Debug` build over our default `Release`.
But there is probably not much harm having it everywhere (just another dependency).
@JafarAbdi could you patch the dependencies everywhere?

PR #1226 (2022-05-10): Make moveit_common a 'depend' rather than 'build_depend'
### Description

Fix: #1094

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit/blob/master/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [reference](https://ros-planning.github.io/moveit_tutorials/doc/tests/tests_tutor [... cut]
closed 2022-05-10 by henningkayser