moveit/moveit2 #1106: setup.py deprecated (launch_param_builder, moveit_config_utils)

henningkayser, 2022-03-08:
### Description

Currently, `launch_param_builder` and `moveit_config_utils` are showing deprecation warnings because of setup.py, recommending to use pip or other standards-based tools instead. 

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source build
* Branch: main

### Backtrace or Console output

```
Starting >>> launch_param_builder
--- stderr: launch_param_builder                                                                  
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_param_builder [0.50s]
Starting >>> moveit_configs_utils
--- stderr: moveit_configs_utils
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [0.44s]
```

schornakj, 2022-04-14:
This issue originates upstream in ROS2 -- there's some discussion about it in [this ament_cmake issue](https://github.com/ament/ament_cmake/issues/382). It sounds like it'll require fixes across several core packages to resolve.

henningkayser, 2023-01-26:
Ok, there is apparently nothing to do on our end. Discussion is continued in https://github.com/colcon/colcon-core/issues/454 and https://github.com/ament/ament_cmake/issues/382

closed 2023-01-26 by henningkayser