moveit/moveit2 #1112: Deprecate getPlanningScene() and the cast operator

AndyZe, 2022-03-10:
As mentioned by v4hn [here](https://github.com/ros-planning/moveit2/pull/1100#pullrequestreview-904301643):

> Additionally I would propose to deprecate the getPlanningScene() method that goes along with these interfaces.
> It's just ludicrous to have a method with its documentation saying [DON'T USE ME](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L189-L190).
> 
> It would also help usability to have a simple getScene() method in the locking classes instead of the [awful cast operator](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L651-L654).

abhijelly, 2022-10-18:
Hi @AndyZe, I cannot seem to find `getPlanningScene()` definition and `getScene()` in the codebase. Is this issue resolved or do I have to look elsewhere? Thank you!

AndyZe, 2022-10-22:
`getPlanningScene()` is definitely still an issue and you can easily find all usages like so:

`grep -r "getPlanningScene("`

I wouldn't worry about `getScene()` for now, let's take care of `getPlanningScene()` first.

Basically this issue asks you to add a deprecation warning to the [function declaration](https://github.com/ros-planning/moveit2/blob/44821258a71c54a1b2381d8e664dfcdac79c40fa/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h#L212) like this:

`[[deprecated("getPlanningScene() is deprecated, use LockedPlanningSceneRO or LockedPlanningSceneRW instead.")]] const planning_scene::PlanningScenePtr& getPlanningScene()`

Since getPlanningScene() is used internally in a bunch of places, we need to replace it somehow. Use LockedPlanningSceneRO or LockedPlanningSceneRW as described in this tutorial:  https://ros-planning.github.io/moveit_tutorials/doc/motion_planning_pipeline/motion_planning_pipeline_tutorial.html#start

@v4hn @tylerjw @henningkayser  does that sound like a good path forward to you? It's going to mean a lot more internal locking of the planning scene.

abhijelly, 2022-10-23:
@AndyZe Thank you for your help! I've raised #1635.