moveit/moveit2 #1120: rosdep cannot find packages on Ubuntu Focal to build moveit2 source

cgarry-vs, 2022-03-14:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 20.04.4 LTS
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? main

### Steps to reproduce
Follow the MoveIt 2 Source Build - Linux guide at https://moveit.ros.org/install-moveit2/source/ and you will fail at the colcon build stage, as no moveit dependencies can be found for Ubuntu Focal (see gist).

### Expected behaviour
To be able to install moveit2 for rosrolling on Ubuntu 20.04.4 Focal LTS

### Actual behaviour
Fails during colcon build step.

### Backtrace or Console output
https://gist.github.com/cgarry-vs/6aa9709ec5b91f0f88f4322f123277c4

tylerjw, 2022-03-14:
Rolling is now on Jammy (22.04) so rosdep has removed all those keys from Focal (20.04).

gavanderhoorn, 2022-03-15:
Just a quick note: the missing packages are not system dependencies, they are ROS packages (there is a difference between these two when it comes to `rosdep` @tylerjw: ROS pkgs don't have "keys" registered).

In order to fix this, you'd have to build all the listed packages from source as well.

You could use `rosinstall_generator` to generate a complete `.repos` file for you, including all the (recursive) dependencies `rosdep` complains about.

cgarry-vs, 2022-03-15:
@tylerjw @gavanderhoorn thanks! what is the supported/recommended setup then? I can make a PR to the docs if building from source is the recommended way, or Ubuntu 20.04 -> Ubuntu 22.04 in the docs.

vatanaksoytezer, 2022-03-15:
Hello @cgarry-vs thanks for offering help. For main branch our CI runs on Galactic right now. Rolling / Jammy was pretty usable before, until Ubuntu 22.04 decided to migrate from Python 3.9 to Python 3.10 and now everything including rclcpp, ros and non-ros stuff is broken, see: https://discourse.ros.org/t/preparing-for-rolling-sync-delayed-by-python-3-10-transition/24521/8. Unfortunately us and all ROS packages are waiting for the Python transition to complete on Jammu, to have Rolling to be usable again.

If you want to use rolling on focal you need to change your ros index url to an older one (`export ROSDISTRO_INDEX_URL=https://raw.githubusercontent.com/ros/rosdistro/rolling/2022-01-28/index-v4.yaml`) and do a source build.

I would be more than happy to review if you open a PR, to update the docs.

cgarry-vs, 2022-03-15:
@gavanderhoorn generated a repos file:

```rosinstall_generator --rosdistro rolling --format repos --upstream-development ALL > moveit2.repos``` and followed the doc instructions side from that. Build got a lot further but still failed.

https://gist.github.com/cgarry-vs/1cdef350d59c91af71bf284d78d24010

gavanderhoorn, 2022-03-15:
Apart from build issues caused by the problems @vatanaksoytezer mentions, you wouldn't want to do a full froms-source build.

You should be able to ask `rosinstall_generator` to skip packages you already have installed (something like `--exclude RPP`, but see the `--help`).

cgarry-vs, 2022-03-15:
@gavanderhoorn thanks. Deleted everything and followed the docs again so pretty sure I had a clean workspace. Will give it another go later though.

aditya2592, 2022-03-22:
> Hello @cgarry-vs thanks for offering help. For main branch our CI runs on Galactic right now. Rolling / Jammy was pretty usable before, until Ubuntu 22.04 decided to migrate from Python 3.9 to Python 3.10 and now everything including rclcpp, ros and non-ros stuff is broken, see: https://discourse.ros.org/t/preparing-for-rolling-sync-delayed-by-python-3-10-transition/24521/8. Unfortunately us and all ROS packages are waiting for the Python transition to complete on Jammy, for Rolling to be usable again.
> 
> For my experience I currently use 20.04 / Galactic as a maintainer to contribute MoveIt.
> 
> If you want to use rolling on focal you need to change your ros index url to an older one (`export ROSDISTRO_INDEX_URL=https://raw.githubusercontent.com/ros/rosdistro/rolling/2022-01-28/index-v4.yaml`) and do a source build.
> 
> I would be more than happy to review if you open a PR, to update the docs.

@vatanaksoytezer How are you setting up moveit with ubuntu 20.04 and galactic? Are we supposed to use the `galactic` branch in this repo? And should we use `rosinstall_generator` with `galactic` as distro or rolling?

vatanaksoytezer, 2022-03-22:
@aditya2592 You can use the main branch and follow the instructions on https://moveit.ros.org/install-moveit2/source/

aditya2592, 2022-03-22:
@vatanaksoytezer Are those instructions also valid when ROS2 is installed from source? I am seeing issues described in this post - https://github.com/ros-planning/moveit2/issues/156 and I was wondering if different instructions are needed

cgarry-vs, 2022-03-23:
Followed the mentioned instructions, but set the ROS index after cloning moveit2/main and the dependencies installed without issue (at least for me on Ubuntu Focal 20.04, ROS Rolling).

```
export ROSDISTRO_INDEX_URL=https://raw.githubusercontent.com/ros/rosdistro/rolling/2022-01-28/index-v4.yaml
```

@vatanaksoytezer https://github.com/ros-planning/moveit.ros.org/pull/702

closed 2022-03-23 by cgarry-vs

cgarry-vs, 2022-10-11:
I've gotten it to work without having to build any dependencies from source
by using the old ros package index recommended in the earlier comment (be
sure to rosdep init, update, etc. in the MoveIt2 repo after the clone step)
and following the rest of the MoveIt2 build instructions as usual.