moveit/moveit2 #1123: TIME!

v4hn, 2022-03-15:
MoveIt's PlanningScene does not have any concept of time. In itself that's ok, because we want to deal with "snapshots" of a real state in many situations. Trajectories already have their own `duration_from_previous` fields, so they are handled externally.

But every time you want to synchronize a planning scene between different components you have to explicitly request it again (unless you have access to the memory of "the one true scene" in `move_group` (or in your `moveit_servo` loop, or any other module you use a `PlanningSceneMonitor` as an authority). It would be useful to be able to state "use a `PlanningScene` from this monitor which is at least as new as <time stamp>`.

The easiest solution I see would be to add a stamp to the `PlanningScene` structure, or at least to the message.
That would be somewhat for C++ trajectories though where scenes can have stamps then but are also associated with durations in the trajectory.

Any other ideas?

henningkayser, 2022-03-18:
> It would be useful to be able to state "use a PlanningScene from this monitor which is at least as new as `.
What would change compared to the current behavior? We don't have a scene buffer or something like that. Do you imply adding a buffer to PSM and maybe make it lock-free with that? ;)

The message has stamps for transforms and the RobotState (hidden inside the JointState header). While I think adding a stamp is trivial and doesn't hurt, I'm not sure I follow the use case with the current implementation of the PSM.

v4hn, 2022-03-19:
> What would change compared to the current behavior?

I'm mostly complaining about a missing interface to tell the PSM to wait until it has a scene with updates until time X. That's required to synchronize scene updates on a move group with external nodes monitoring that scene. 
the joint state message is not necessarily a workaround because it's time stamp implies the *last joint state motion*, not the current time of the scene.

it's not an urgent issue for me, but I run into it every now and then.