moveit/moveit2 #1126: occupancy map server crash when is executed

JeisonUR, 2022-03-17:
### Description
the node throw instance of 'std::runtime_error' saying that the Node has already been added to an executor.

### Your environment
* ROS Distro:
* OS Version: e.g. Ubuntu 18.04
* Source 
* branch main

### Backtrace or Console output
```sh
ros2 run moveit_ros_occupancy_map_monitor moveit_ros_occupancy_map_server 
```
[WARN] [1647551938.090227400] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[ERROR] [1647551938.090324926] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 'octomap_frame' parameter defined for octomap updates
[ERROR] [1647551938.090349020] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
[WARN] [1647551938.090367319] [moveit.ros.occupancy_map_monitor]: No target frame specified for Octomap. No transforms will be applied to received data.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Node has already been added to an executor.
Stack trace (most recent call last):
#12   Object "", at 0xffffffffffffffff, in 
#11   Object "/home/jeisonur/ws_moveit2/install/moveit_ros_occupancy_map_monitor/lib/moveit_ros_occupancy_map_monitor/moveit_ros_occupancy_map_server", at 0x556e53f4537d, in _start
#10   Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7f9c3d1b80b2]
#9    Object "/home/jeisonur/ws_moveit2/install/moveit_ros_occupancy_map_monitor/lib/moveit_ros_occupancy_map_monitor/moveit_ros_occupancy_map_server", at 0x556e53f44f95, in main
#8    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f9c3d818f8f, in rclcpp::spin(std::shared_ptr<rclcpp::Node>)
#7    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f9c3d818ca8, in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
#6    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7f9c3d7ecc66, in 
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f9c3d44b6a8, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f9c3d44b3f6, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f9c3d44b38b, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f9c3d43f910, in 
#1    Source "/build/glibc-sMfBJT/glibc-2.31/stdlib/abort.c", line 79, in abort [0x7f9c3d1b6858]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7f9c3d1d703b]

henningkayser, 2022-03-18:
@JeisonUR thanks for reporting. Let me know if #1130 fixes this for you.

henningkayser, 2022-04-21:
#1130 fixes this from what I've seen

closed 2022-04-21 by henningkayser