moveit/moveit2 #1143: Commit 8ed8 broke the Motion Planning RViz plugin

AndyZe, 2022-03-27:
### Description

[This commit ](https://github.com/ros-planning/moveit2/commit/8ed8c1b45ed8e70dbb892788fcb9a36581624973) seems to have broken the Motion Planning RViz plugin. I traced it down to this commit by stepping through the commit history one by one.

### Your environment
* ROS Distro: ROS2 Rolling
* OS Version: Ubuntu 20.04
* Source build of ROS2 branches

### Steps to reproduce
Launch the Panda demo:

`ros2 launch moveit_resources_panda_moveit_config demo.launch.py`

When RViz opens, try to add the motion planning plugin. For me, this is the terminal output. (Note that this warning is normal. At least it was occurring prior this bug without affecting functionality:  `Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class `)

```
[rviz2-1] [INFO] [1648422303.707767300] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1648422303.708609614] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-1]          at line 253 in /opt/ros/rolling/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-1] [ERROR] [1648422312.584880710] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1648422312.598373754] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [INFO] [1648422312.665576616] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1648422312.666456537] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-1] [INFO] [1648422312.671458345] [interactive_marker_display_94111531476016]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-1] [INFO] [1648422312.673906341] [moveit_ros_robot_interaction.robot_interaction]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-1] Link panda_link1 had 1 children
...
[rviz2-1] Link panda_rightfinger had 0 children
[rviz2-1] [INFO] [1648422312.674323650] [moveit_ros_robot_interaction.robot_interaction]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-1] /opt/ros/rolling/lib/rviz2/rviz2: symbol lookup error: /home/andy/ws_ros2/install/moveit_ros_visualization/lib/libmoveit_motion_planning_rviz_plugin_core.so.2.4.0: undefined symbol: _ZNK14planning_scene13PlanningScene17getCollidingPairsERSt3mapISt4pairINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_ESt6vectorIN19collision_detection7ContactESaISC_EESt4lessIS9_ESaIS2_IKS9_SE_EEERKN6moveit4core10RobotStateERKNSB_22AllowedCollisionMatrixE
[ERROR] [rviz2-1]: process has died [pid 270608, exit code 127, cmd '/opt/ros/rolling/lib/rviz2/rviz2 -d /home/andy/ws_ros2/install/moveit_resources_panda_moveit_config/share/moveit_resources_panda_moveit_config/launch/moveit.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_gjpve4gm --params-file /tmp/launch_params_sxr_ae5r --params-file /tmp/launch_params_u2ff4b1g --params-file /tmp/launch_params_i2xyg9xe'].
```

AndyZe, 2022-03-28:
@Abishalini do you have a chance to look at this? We don't have adequate tests to catch this kind of bug  :(

AndyZe, 2022-03-29:
The issue is gone since I purged moveit binaries and updated the ros2_control packages (which I was building from source). Thanks @Abishalini

closed 2022-03-29 by AndyZe

prasuchit, 2024-04-11:
@AndyZe  @Abishalini I'm facing the same error with[ KUKA lbr iisy moveit packages](https://github.com/kroshu/kuka_drivers/blob/master/examples/iiqka_moveit_example/launch/moveit_planning_example.launch.py). I also cannot see the Moveit interactive marker (the blue sphere) around the endeffector.

Note: I'm on ROS2 Humble and all my packages and dependencies are up to date.

Are these two things connected? How do I fix them? Could you point me in the right direction?