moveit/moveit2 #1159: Pick and place functions

Jeisongarcia9713, 2022-04-01:
I am working with moveit2 and I need to implement pick and place, but I see that the code for pick and place is commented in the planning interface, I wanna know if the function is disabled and now there is another approach to implement this, or maybe is disabled because is in process of migration to ros2?

AndyZe, 2022-04-01:
MoveIt Task Constructor is probably your best option. They have a pick/place demo here:

https://github.com/ros-planning/moveit_task_constructor/tree/ros2/demo

Jeisongarcia9713, 2022-04-01:
yes, I have tried with that too but for some reason, the demo failed in the planning, even though I am going to check out again and try with that, thanks. But that means that by now is disabled or is permanent? just curious about it

AndyZe, 2022-04-01:
In my opinion, the MoveIt pick/place interface won't be coming back

JafarAbdi, 2022-04-04:
As @AndyZe the plan is to remove it from moveit2 in favor of MTC

closed 2022-04-04 by JafarAbdi