moveit/moveit2 #1164: Add mesh file to planning scene as collision object

beta1scat, 2022-04-04:
### Description
I want to add mesh file (e.g. *.stl, *.obj, *.dae) to planning scene as collision object, but rviz2 print a lot of error messages about materials.
My code is :
```
moveit_msgs::msg::CollisionObject collision_object2;
    collision_object2.header.frame_id = "base_link";
    collision_object2.id = "case";

    shapes::Mesh * m = shapes::createMeshFromResource("package://tf_pick/meshes/untitled.dae");
    // /home/niu/ws_ros2/src/tf_pick/meshes/ciif-case.obj
    shape_msgs::msg::Mesh shelf_mesh;
    shapes::ShapeMsg shelf_mesh_msg;
    shapes::constructMsgFromShape(m,shelf_mesh_msg);
    shelf_mesh = boost::get<shape_msgs::msg::Mesh>(shelf_mesh_msg);

    /* A pose for the box (specified relative to frame_id) */
    geometry_msgs::msg::Pose shelf_pose;
    shelf_pose.orientation.w = 1;
    shelf_pose.orientation.x = 0;
    shelf_pose.orientation.y = 0;
    shelf_pose.orientation.z = 0;
    shelf_pose.position.x =  0.3;
    shelf_pose.position.y =  -0.25;
    shelf_pose.position.z =  -0.1;

    collision_object2.meshes.push_back(shelf_mesh);
    collision_object2.mesh_poses.push_back(shelf_pose);
    collision_object2.operation = collision_object2.ADD;
    // Add object to planning scene
    {  // Lock PlanningScene
      planning_scene_monitor::LockedPlanningSceneRW scene(this->moveit_cpp_->getPlanningSceneMonitor());
      scene->processCollisionObjectMsg(collision_object2);
    }  // Unlock PlanningScene
```
Results:
```
[rviz2-4] [ERROR] [1649066982.936072797] [rviz2]: Can't assign material Shape22Material to the ManualObject MeshShape_ManualObject0 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-4] [ERROR] [1649066983.432405386] [rviz2]: Can't assign material Shape24Material to the ManualObject MeshShape_ManualObject1 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-4] [ERROR] [1649066983.926929507] [rviz2]: Can't assign material Shape26Material to the ManualObject MeshShape_ManualObject2 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-4] [ERROR] [1649066984.440845637] [rviz2]: Can't assign material Shape28Material to the ManualObject MeshShape_ManualObject3 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-4] [ERROR] [1649066984.953925202] [rviz2]: Can't assign material Shape30Material to the ManualObject MeshShape_ManualObject4 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-4] [ERROR] [1649066985.436153708] [rviz2]: Can't assign material Shape32Material to the ManualObject MeshShape_ManualObject5 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
[rviz2-4] [ERROR] [1649066985.953075143] [rviz2]: Can't assign material Shape34Material to the ManualObject MeshShape_ManualObject6 because this Material does not exist in group General. Have you forgotten to define it in a .material script?
```

### Your environment
* ROS Distro: [Foxy]
* OS Version: e.g. Ubuntu 20.04
* ROS install: apt
* moveit2 install: source build, branch foxy

JafarAbdi, 2022-04-04:
Hi @beta1scat, can you check if https://github.com/ros-planning/moveit2/pull/672 fixes the issue for you? I'll cherry-pick it back to foxy if it does (not sure if it's already there)

beta1scat, 2022-04-05:
> 

Hi @JafarAbdi, I have tried to checkout to branch `5099f70001762e5f8ec5dbde84b6f6ff3728274d`(#672), it doesn't work. I found the changes in #672 at two positions. So I changed the code at branch `foxy` manually. It works! Thank you very much!

closed 2022-04-06 by beta1scat