moveit/moveit2 #1175: Hybrid planning action types are not flexible, require re-time-parameterization

AndyZe, 2022-04-12:
Hybrid planning is designed to be flexible, with local planner and global planner plugins. The plugins exchange these message types:

moveit_msgs::action::GlobalPlanner
moveit_msgs::action::LocalPlanner
moveit_msgs::action::HybridPlanner

These actions don't contain trajectory information. They only contain a [MotionPlanRequestMsg](https://github.com/ros-planning/moveit_msgs/blob/ros2/msg/MotionPlanRequest.msg). I don't think this is ideal because it requires time-parameterization to be re-run. In other words, the local planner doesn't have any idea of _when_ it should reach the next waypoint.

I think it would be nice to replace the `MotionPlanRequest[]` with `trajectory_msgs/JointTrajectory[]` so that timing information is included.

sjahr, 2022-06-02:
@AndyZe I am a bit confused what you mean with this issue. Between the Global and the Local Planner, time information is exchanged through the `global_trajectory` topic. The message type of this topic is `MotionPlanResponse` which contains a field `RobotTrajectory` that contains a `trajectory_msgs/JointTrajectory`. So from my understanding, the local planner receives a fully parameterized trajectory from the global planner.

sjahr, 2023-11-28:
Closing this because I think it is outdated. Also the MotionPlanRequest message contains a field called reference trajectory that can store a JointTrajectory. Passing the full Request is necessary to also provide information about constraints. @AndyZe Feel free to re-open this issue if I am missing something here

closed 2023-11-28 by sjahr