moveit/moveit2 #1182: How to specify a controller to use when executing a trajectory through the MoveGroup ROS interface?

schornakj, 2022-04-14:
When the ExecuteTrajectoryAction MoveGroup capability [pushes a new trajectory](https://github.com/ros-planning/moveit2/blob/14d2a932c74ab1dc6bed652738d17ee2bae23d4f/moveit_ros/move_group/src/default_capabilities/execute_trajectory_action_capability.cpp#L111) to the TrajectoryExecutionManager, it does not have a way to set what controller should be used to execute the trajectory. This results in the TrajectoryExecutionManager being left to [figure out what controller should be used](https://github.com/ros-planning/moveit2/blob/14d2a932c74ab1dc6bed652738d17ee2bae23d4f/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L728-L810). In situations where there are multiple possible controllers for the same set of joints (for example, a JointGroupPositionController used for Servo and a JointTrajectoryController for regular trajectory execution), the TEM simply doesn't have enough contextual info to make the right decision, which results in undesirable results if the wrong controller is selected.

I think the solution to this is to add a field for controller names to the [ExecuteTrajectory.action message](https://github.com/ros-planning/moveit_msgs/blob/ros2/action/ExecuteTrajectory.action) to allow the code on the action client side to set the right controllers when the motion is planned. This would require an API change, though.

Does anyone have other ideas about this?

AndyZe, 2022-04-14:
IIRC you can specify a default controller name. For example:

https://github.com/ros-planning/moveit_resources/blob/394a3c6aba9da34d5317697fdb0db2c9acc8ebd8/panda_moveit_config/config/moveit_controllers.yaml#L10

Just turn off `moveit_manage_controllers` so trajectories execute with that controller by default, instead of searching for an active controller.

Servo can have a different controller topic and that should not affect things.

AndyZe, 2022-04-14:
I think the controller search logic should just be deleted. If you try what I wrote above and it still doesn't work, then I support your idea  :+1:

schornakj, 2022-04-14:
> Just turn off moveit_manage_controllers so trajectories execute with that controller by default, instead of searching for an active controller.

Setting `moveit_manage_controllers = False` would require me to switch between the controllers elsewhere in my code, right? If the controllers aren't switched automatically before handling the trajectory, I would have to make sure that the streaming controller is inactive before executing a trajectory that should use the joint trajectory controller.

I think that the TEM actually does not have a way to read if a given controller is set as default or not. I see that this part of the controller state is read [here](https://github.com/ros-planning/moveit2/blob/14d2a932c74ab1dc6bed652738d17ee2bae23d4f/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L778-L779), but I can't find where we read that parameter and set its value in the controller state info.

AndyZe, 2022-04-14:
> I would have to make sure that the streaming controller is inactive before executing a trajectory that should use the joint trajectory controller.

The ros2_control resource manager should prevent that.

I don't want to sound too negative about your idea. If we add this new feature, can we strip out some "controller searching" code, too? Don't want to keep adding and adding without removing.

schornakj, 2022-04-14:
> The ros2_control resource manager should prevent that.

What do you mean when you say "prevent"? I should expand a bit on what I posted earlier to be clearer:

If the controllers aren't switched automatically before handling the trajectory **and I want to successfully execute a trajectory using the joint trajectory controller even if the streaming controller is currently active**, I would have to make sure that the streaming controller is inactive before executing a trajectory that should use the joint trajectory controller **because the controllers will no longer be switched automatically if the current configuration of the system is different from the configuration needed to execute the trajectory with the desired controller**.

I agree that the TEM's controller search logic should be largely removed. It's kind of broken anyway. A minor complication is that lots of code that uses MoveIt assumes that if no particular controller is specified when executing a trajectory then the TEM will pick the right controller to use. Since most projects just use one controller per set of joints, it's probably acceptable to replace the TEM's controller search logic with some simple condition, like picking the first currently-active controller that controls the joints actuated by the trajectory.

AndyZe, 2022-04-14:
> Since most projects just use one controller per set of joints, it's probably acceptable to replace the TEM's controller search logic with some simple condition, like picking the first currently-active controller that controls the joints actuated by the trajectory.

I'd prefer something even simpler: just take what's in the [yaml file](https://github.com/ros-planning/moveit_resources/blob/394a3c6aba9da34d5317697fdb0db2c9acc8ebd8/panda_moveit_config/config/moveit_controllers.yaml#L10) as the default. If the default is not active, then so be it. MoveIt shouldn't be searching.

schornakj, 2022-04-14:
> I'd prefer something even simpler: just take what's in the [yaml [...]

One issue: this YAML file is only used by the [MoveItSimpleControllerManager](https://github.com/ros-planning/moveit2/blob/main/moveit_plugins/moveit_simple_controller_manager/src/moveit_simple_controller_manager.cpp), right? I'm using the [MoveItControllerManager](https://github.com/ros-planning/moveit2/blob/main/moveit_plugins/moveit_ros_control_interface/src/controller_manager_plugin.cpp), which gets controller states from the ros2_control ControllerManager, and ros2_control does not have an equivalent concept of a controller being set as a default.

TomCC7, 2024-06-04:
Hello, I'd like to follow up on this issue. I noticed that the mechanism to provide controllers for a specific trajectory execution to the TEM was already implemented [here](https://github.com/moveit/moveit2/blob/672e0ecd48f471fe05057f81e6233272a692a156/moveit_ros/planning/trajectory_execution_manager/include/moveit/trajectory_execution_manager/trajectory_execution_manager.h#L147). I didn't find where the function is being used, but if it's already there, why not just adding that into the action parameter also? What's your concern?

Cody-Vu, 2025-04-10:
Hello, I am using the moveit simple controller manager. Is it possible to configure multiple controllers for the same set of joints and switch the controllers during runtime? I tried but MoveIt only sends the trajectories to the controller which is set as default. But I want it to send it to the one which is currently running. Appreciate any further information!