moveit/moveit2 #1188: Rolling Jammy Buildfarm Failures

vatanaksoytezer, 2022-04-19:
The following packages have been failing in Rolling Jammy buildfarm:

```
moveit_visual_tools
warehouse_ros_sqlite
moveit2
```

Also warehouse_ros_mongo is failing, but as mongo is no longer in Ubuntu 22.04, we should consider dropping it.

ijnek, 2022-04-27:
It seems like most changes to compile on Ubuntu Jammy have been completed in https://github.com/ros-planning/moveit2/pull/1095 as mentioned in https://github.com/ros-planning/moveit2/issues/1092 but a rolling release hasn't been made yet.

It seems like a [rolling release for srdfdom](https://github.com/ros-planning/srdfdom/issues/106) would have to happen before moveit2 can be released, but the rest of moveit2 seems to compile fine from source.

v4hn, 2022-05-19:
@JafarAbdi some people from my lab are working on jammy migration for their stack and would really appreciate a release.
Could you or @vatanaksoytezer manage any time soon? :-)

JafarAbdi, 2022-05-19:
> @JafarAbdi some people from my lab are working on jammy migration for their stack and would really appreciate a release. Could you or @vatanaksoytezer manage any time soon? :-)

We're waiting on https://github.com/ros-planning/moveit2/pull/1254 to cut a new one, I'll release it to Rolling as soon as we release moveit to Humble

v4hn, 2022-05-19:
I don't know what you're timeline for merging the feature branch is @DLu , but @henningkayser's review mentions multiple issues and I guess paid projects and family life take some priority over finalizing the pr.

Unless it is critical not to release before David's great work is merged (from what I looked into it's not, but I am not directly involved) I would prefer if you could cut a release this weekend.
Please note this is rolling, so major changes *can and should* be merged there when they are ready independent of the currently targeted OS distribution release.

DLu, 2022-05-19:
I've been cramming the past few days to address the concerns. I believe its all set.

v4hn, 2022-05-25:
> I would prefer if you could cut a release this weekend. [...] Please note this is rolling, so major changes can and should be merged there when they are ready independent of the currently targeted OS distribution release.

@vatanaksoytezer @JafarAbdi Please discuss at the standup today (if there is one?) I don't think it's necessary to wait with the release. The next release will of course improve a lot either way :-)

vatanaksoytezer, 2022-05-26:
https://github.com/ros/rosdistro/pull/33377 release has been cut for rolling

henningkayser, 2022-06-02:
all green on buildfarm :heavy_check_mark: , let's close this

closed 2022-06-02 by henningkayser