moveit/moveit2 #1190: Use SensorDataQoS() when subscribing to joint_states

smith-doug, 2022-04-20:
### Description

It looks like the joint_states topic can/will be published as rclcpp::SensorDataQoS() in Humble.  The [current_state_monitor](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor_middleware_handle.cpp#L65) does not use SensorDataQoS(), and will not be able to subscribe to joint_states if it's published as sensor data.  I am working with a new in-development robot driver that publishes joint_states as sensor data.

More info: https://answers.ros.org/question/399325/qos-settings-for-joint_states-in-humble/   

I don't know if the correct behavior is already defined in a REP somewhere.  I would recommend defaulting to sensor data and making it configurable, like in [robot_state_publisher](https://github.com/ros/robot_state_publisher/blob/ros2/src/robot_state_publisher.cpp#L139).

### Your environment
* ROS Distro: galactic
* OS Version:  Ubuntu 20.04
* Binary and Source (galactic and main)

### Steps to reproduce
Publish joint_states with a SensorDataQoS publisher and try to get the robot's current state with the planning_scene_monitor/current_state_monitor.

### Expected behaviour
joint_states should be received and current_state_monitor should work.

### Actual behaviour
joint_states is never received and the planning_scene_monitor times out.

tylerjw, 2022-04-26:
To respond to this with an extra data point. All of the robots we are currently using with MoveIt2 use the joint_state_broadcaster from ros2_controllers which uses SystemDefaultsQoS

Here is the line where it creates the publisher: https://github.com/ros-controls/ros2_controllers/blob/7acf29c191ea44870c7327436a4579e06261fde9/joint_state_broadcaster/src/joint_state_broadcaster.cpp#L156

I don't really understand why you would want to use one or the other or what robot_state_publisher is and why you would use it over ros2_control.

What are the implications of SensorDataQoS vs SystemDefaultQoS? Is there a reason one is more appropriate for joint_states topics than the other? If so we would need to coordinate also changing ros2_control's use of this topic.

gavanderhoorn, 2022-04-26:
> I don't really understand [..] what robot_state_publisher is and why you would use it over ros2_control.

RSP performs FK based on incoming `JointState` messages and the URDF, to produce the relevant TF messages on `/tf` and `/tf_static`.

`ros2_control` does not have anything equivalent.

nbbrooks, 2022-04-26:
I made a lightnin talk on my DDS/QOS learnings for the February MoveIt Community meeting. You can see the full recording [here](https://www.youtube.com/watch?v=WajZmurQYMc&ab_channel=NathanBrooks).

Here is a table with some ideas I've had around out of box QOS profiles that could be helpful. Items in green are proposed new QOS (I renamed "sensor data" to "low frequency data").
[image]

In practice, I think it is common to use "best effort" for high frequency data, such as joint states. I have not had time to benchmark anything on hardware though so I do not know if there is a substantial impact on performance changing to it, nor if this [interesting edge case](https://www.rti.com/blog/2012/05/24/reliability-isnt-just-for-getting-everything-that-was-sent/) is encountered.

nbbrooks, 2022-04-26:
> It looks like the joint_states topic can/will be published as rclcpp::SensorDataQoS() in Humble
Could you provide a link to this? I've only seen this [RSP PR](https://github.com/ros/robot_state_publisher/pull/179), which just changed its *subscription* to use SensorData by default.

smith-doug, 2022-04-27:
I updated my [post on ROS Answers ](https://answers.ros.org/question/399325/qos-settings-for-joint_states-in-humble/?answer=399379#post-id-399379) with the feedback I got during a call with the ROS2 TSC. 

I interpreted them changing the subscription type in a ros_base package to mean that "publishers and subscribers should use best_effort reliability for joint_states".  I was also working with a new experimental robot driver that published joint_states as best_effort.  

In reality, their decision was more like "Let's make robot_state_publisher's **subscriber** use best_effort, since that can receive reliable and best_effort messages, and make it reconfigurable."  There isn't any official guideline for publishers of this topic.  I do think that their use of best_effort subscriber in a ros_base package is a fair enough guideline to do the same.

I'm not an authority on the subject, but I don't think you have to change the publisher in ros2_controller.  It would be nice if it was configurable.  The burden falls on the subscriber to make sure it has a compatible reliability.  A best_effort subscriber should be able to receive both reliable and best_effort messages fine.  Making it configurable like RSP did would satisfy people that are worried about only receiving from reliable publishers.

nbbrooks, 2022-04-27:
Understood, so this is about future proofing CSM by making it subscribe using "Best Effort", since Best Effort subscribers can receive from both "Best Effort" and "Reliable" publishers. Whether joint state publishers should switch from Reliable to Best Effort is a separate topic (but we should at least make sure they are configurable from the launch files).

AndyZe, 2022-04-27:
This makes sense to me.

AndyZe, 2022-05-17:
> It looks like the joint_states topic can/will be published as rclcpp::SensorDataQoS() in Humble. The [...]

I don't believe that's true. I believe it will still be able to subscribe, per the table [here](https://docs.ros.org/en/rolling/Concepts/About-Quality-of-Service-Settings.html). Either way, I support this idea.