moveit/moveit2 #1192: Quickstart broken

odellus, 2022-04-22:
### Description

The quickstart doesn't work. I've tried the fix suggested in #826 and it isn't working for me.

### Your environment
* ROS Distro: galactic
* OS Version: Ubuntu 20.04
* I'm following the quickstart.

### Steps to reproduce
Try to get the quickstart to work.

### Expected behaviour
The quickstart should work

### Actual behaviour
The quickstart doesn't work

### Backtrace or Console output

```
/home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
/home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:76: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:358: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit2_tutorials [32.8s, exited with code 2]
                                     
Summary: 8 packages finished [1min 2s]
  1 package failed: moveit2_tutorials
  2 packages had stderr output: moveit2_tutorials moveit_visual_tools
```

This issue is an exact duplicate of #826 . The fix proposed there does not work for me.

odellus, 2022-04-22:
One inconsistency I've noticed is that the beginning of the quickstart says
```
source /opt/ros/galactic/setup.bash
```
but the suggested fix implies that finding packages in `/opt` is the problem.

I reinstalled `ros2-galactic` with the instructions found [here](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Binary.html). Omitting the 
```
source /opt/ros/galactic/setup.bash
```
gives me 
```
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findmoveit_common.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "moveit_common", but CMake did not find one.

  Could not find a package configuration file provided by "moveit_common"
  with any of the following names:

    moveit_commonConfig.cmake
    moveit_common-config.cmake

  Add the installation prefix of "moveit_common" to CMAKE_PREFIX_PATH or set
  "moveit_common_DIR" to a directory containing one of the above files.  If
  "moveit_common" provides a separate development package or SDK, be sure it
  has been installed.
```

I installed ros2 [from source](https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html) after `apt purge ros-*` and even with doing that it seems like `rosdep install -r` is installing everything to `/opt`.

Without the call to source the setup in `/opt`, I still see the same error as before. After sourceing and re-running
```
cd ~/ws_moveit2
colcon build --mixin release
```
I get the following [error](https://gist.github.com/odellus/22e6523ab91b3faf881c477235f8b948) which shows things like 
```
dir 0 is [/opt/ros/galactic/lib]
dir 1 must precede it due to runtime library [librcl_interfaces__rosidl_generator_c.so]
...
dir 1 is [/home/thomas/ros2_galactic/install/rcl_interfaces/lib]
      dir 0 must precede it due to runtime library [librcl_interfaces__rosidl_generator_c.so]
```
so I'm going to try to build moveit2 from source without any vague `rosdep install -r` calls to see if I can get this quickstart running today.

odellus, 2022-04-22:
Attempting to install moveit2 from source with a ro2 built from source results in errors identical to #156. Trying the solution suggested there to install ro2 from binaries before attempting another build of moveit2.

odellus, 2022-04-22:
Installing ros2 from binaries and attempting to build moveit2 from source gives the same error. I did this not long ago. What happend? When did everything break?
```
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimpTargets.cmake:54 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimp-config.cmake:1 (include)
  CMakeLists.txt:41 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:44 (find_package):
  By not providing "Findeigen_stl_containers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "eigen_stl_containers", but CMake did not find one.

  Could not find a package configuration file provided by
  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

  Add the installation prefix of "eigen_stl_containers" to CMAKE_PREFIX_PATH
  or set "eigen_stl_containers_DIR" to a directory containing one of the
  above files.  If "eigen_stl_containers" provides a separate development
  package or SDK, be sure it has been installed.
```
Tried to manually install eigen-stl-containers
```
sudo apt install ros-galactic-eigen-stl-containers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-galactic-eigen-stl-containers is already the newest version (1.0.0-3focal.20220124.195811).
```

odellus, 2022-04-23:
I was able to build MoveIt2! from source with this Dockerfile. Closing. 🤬 
```Dockerfile
FROM ubuntu:20.04

WORKDIR /app

# setup locale
RUN apt-get update && \
    apt-get install locales && \
    locale-gen en_US en_US.UTF-8 && \
    update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && \
    # set default sh to bash
    ln -sf bash /bin/sh

# add repo to package source
RUN apt-get update && \
    export LANG=en_US.UTF-8 && \
    apt-get install -y software-properties-common && \
    add-apt-repository universe && \
    apt-get update && \
    apt-get install -y curl gnupg lsb-release && \
    curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
    echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# install prereqs
RUN apt-get update && \
    apt-get install -y \
    build-essential \
    cmake \
    git \
    libbullet-dev \
    python3-colcon-common-extensions \
    python3-flake8 \
    python3-pip \
    python3-pytest-cov \
    python3-rosdep \
    python3-setuptools \
    python3-vcstool \
    wget \
    clang-format-10 && \
    # install some pip packages needed for testing
    python3 -m pip install -U \
    argcomplete \
    flake8-blind-except \
    flake8-builtins \
    flake8-class-newline \
    flake8-comprehensions \
    flake8-deprecated \
    flake8-docstrings \
    flake8-import-order \
    flake8-quotes \
    pytest-repeat \
    pytest-rerunfailures \
    pytest

# install ros2 galactic
RUN apt-get update && \
    apt-get install -y ros-galactic-desktop

# build moveit2
RUN apt-get update && \
    export LANG=en_US.UTF-8 && \
    . /opt/ros/galactic/setup.bash && \
    rosdep init && \
    rosdep update && \
    export COLCON_WS=/app/ws_moveit2 && \
    mkdir -p $COLCON_WS/src && \
    cd $COLCON_WS/src && \
    git clone https://github.com/ros-planning/moveit2.git -b main && \
    for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done && \
    rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y && \
    cd $COLCON_WS && \
    colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

```

odellus, 2022-04-23:
Okay so I could build MoveIt2! from source following the instructions in [this repo](https://github.com/odellus/moveit2-source-build), but going through the steps in the quickstart to build moveit2_tutorials
```bash
sudo apt install python3-colcon-mixin
export COLCON_WS=~/ws_moveit2
. $COLCON_WS/install/setup.bash
colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml
colcon mixin update default
cd $COLCON_WS/src
git clone https://github.com/ros-planning/moveit2_tutorials.git
vcs import < moveit2_tutorials/moveit2_tutorials.repos
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
cd $COLCON_WS
colcon build --mixin release
```
and I'm still seeing
```
/home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
/home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:76: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:358: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit2_tutorials [38.1s, exited with code 2]

Summary: 43 packages finished [1min 29s]
  1 package failed: moveit2_tutorials
  2 packages had stderr output: moveit2_tutorials moveit_visual_tools
```
so I'm unable to build the tutorials even with building moveit2 from source.

I have been able to reproduce the issue by building this Dockerfile
```Dockerfile
FROM ubuntu:20.04

WORKDIR /app

# setup locale
RUN apt-get update && \
    apt-get install locales && \
    locale-gen en_US en_US.UTF-8 && \
    update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && \
    # set default sh to bash
    ln -sf bash /bin/sh

# add repo to package source
RUN apt-get update && \
    export LANG=en_US.UTF-8 && \
    apt-get install -y software-properties-common && \
    add-apt-repository universe && \
    apt-get update && \
    apt-get install -y curl gnupg lsb-release && \
    curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
    echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# install prereqs
RUN apt-get update && \
    apt-get install -y \
    build-essential \
    cmake \
    git \
    libbullet-dev \
    python3-colcon-common-extensions \
    python3-flake8 \
    python3-pip \
    python3-pytest-cov \
    python3-rosdep \
    python3-setuptools \
    python3-vcstool \
    wget \
    clang-format-10 && \
    # install some pip packages needed for testing
    python3 -m pip install -U \
    argcomplete \
    flake8-blind-except \
    flake8-builtins \
    flake8-class-newline \
    flake8-comprehensions \
    flake8-deprecated \
    flake8-docstrings \
    flake8-import-order \
    flake8-quotes \
    pytest-repeat \
    pytest-rerunfailures \
    pytest

# install ros2 galactic
RUN apt-get update && \
    apt-get install -y ros-galactic-desktop

# build moveit2
RUN apt-get update && \
    export LANG=en_US.UTF-8 && \
    . /opt/ros/galactic/setup.bash && \
    rosdep init && \
    rosdep update && \
    export COLCON_WS=/app/ws_moveit2 && \
    mkdir -p $COLCON_WS/src && \
    cd $COLCON_WS/src && \
    git clone https://github.com/ros-planning/moveit2.git -b main && \
    for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done && \
    rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y && \
    cd $COLCON_WS && \
    colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

RUN apt-get update && \
    export LANG=en_US.UTF-8 && \
    . /opt/ros/galactic/setup.bash

# build moveit2_tutorials
RUN apt-get update && \
    export COLCON_WS=/app/ws_moveit2 && \
    . $COLCON_WS/install/setup.bash && \
    apt-get install -y python3-colcon-mixin && \
    colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml && \
    colcon mixin update default && \
    cd $COLCON_WS/src && \
    git clone https://github.com/ros-planning/moveit2_tutorials.git && \
    vcs import < moveit2_tutorials/moveit2_tutorials.repos && \
    rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y && \
    cd $COLCON_WS && \
    colcon build --mixin release
```

odellus, 2022-04-26:
Any help would be greatly appreciated. Really want to get started using MoveIt2!

AndyZe, 2022-04-26:
One thing you could try is to purge your moveit binaries then build MoveIt from source. ROS2 seems to be having some issues with mixing of debians & source.

`sudo apt remove ros-galactic-moveit*`

Obviously make sure you are on the galactic branch of moveit.

It looks like CI passed as of 5 days ago for the Galactic branch.

odellus, 2022-04-26:
I'm building Moveit2! from source inside a docker container. I even tried building ROS2 from source. I can build Moveit2!, but not the tutorials.

> Okay so I could build MoveIt2! from source following the instructions in [this repo](https://github.com/odellus/moveit2-source-build), but going through the steps in the quickstart to build moveit2_tutorials
> 
> ```shell
> sudo apt install python3-colcon-mixin
> export COLCON_WS=~/ws_moveit2
> . $COLCON_WS/install/setup.bash
> colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml
> colcon mixin update default
> cd $COLCON_WS/src
> git clone https://github.com/ros-planning/moveit2_tutorials.git
> vcs import < moveit2_tutorials/moveit2_tutorials.repos
> rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
> cd $COLCON_WS
> colcon build --mixin release
> ```
> 
> and I'm still seeing
> 
> ```
> /home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
> /home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
>   164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
>       |                                                                             ^~~~~~
> make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:76: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:358: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> make: *** [Makefile:146: all] Error 2
> ---
> Failed   <<< moveit2_tutorials [38.1s, exited with code 2]
> 
> Summary: 43 packages finished [1min 29s]
>   1 package failed: moveit2_tutorials
>   2 packages had stderr output: moveit2_tutorials moveit_visual_tools
> ```
> 
> so I'm unable to build the tutorials even with building moveit2 from source.
> 
> I have been able to reproduce the issue by building this Dockerfile
> 
> ```dockerfile
> FROM ubuntu:20.04
> 
> WORKDIR /app
> 
> # setup locale
> RUN apt-get update && \
>     apt-get install locales && \
>     locale-gen en_US en_US.UTF-8 && \
>     update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && \
>     # set default sh to bash
>     ln -sf bash /bin/sh
> 
> 
> # add repo to package source
> RUN apt-get update && \
>     export LANG=en_US.UTF-8 && \
>     apt-get install -y software-properties-common && \
>     add-apt-repository universe && \
>     apt-get update && \
>     apt-get install -y curl gnupg lsb-release && \
>     curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
>     echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null
> 
> # install prereqs
> RUN apt-get update && \
>     apt-get install -y \
>     build-essential \
>     cmake \
[... 41 lines cut ...]
>     rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y && \
>     cd $COLCON_WS && \
>     colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
> 
> # build moveit2_tutorials
> RUN apt-get update && \
>     export LANG=en_US.UTF-8 && \
>     export COLCON_WS=/app/ws_moveit2 && \
>     . $COLCON_WS/install/setup.bash && \
>     apt-get install -y python3-colcon-mixin && \
>     colcon mixin add default https://raw.githubusercontent.com/colcon/colcon-mixin-repository/master/index.yaml && \
>     colcon mixin update default && \
>     cd $COLCON_WS/src && \
>     git clone https://github.com/ros-planning/moveit2_tutorials.git && \
>     vcs import < moveit2_tutorials/moveit2_tutorials.repos && \
>     rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y && \
>     cd $COLCON_WS && \
>     colcon build --mixin release
> ```

What happened when tried building ROS2 from source.

> Attempting to install moveit2 from source with a ros2 built from source results in errors identical to #156. Trying the solution suggested there to install ros2 from binaries before attempting another build of moveit2.

So should I `apt-get purge ros2-galactic-moveit*` inside the docker container? Does anyone have a containerized version of the tutorial so I can look at their Dockerfile?

odellus, 2022-04-26:
In my original comment
> ```
> rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
> ```
> 
> just reinstalls the packages that
> 
> ```
> sudo apt purge "ros-${ROS_DISTRO}-moveit-*"
> ```
> 
> purges and I'm right back with the same error message I started with.

AndyZe, 2022-04-26:
Sorry this is difficult. I think you should run this:

`rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y`

Then run:

`sudo apt remove ros-${ROS_DISTRO}-moveit*`

If anybody has a docker image you could use, it would probably be @vatanaksoytezer

odellus, 2022-04-26:
Alright! That helps a lot. I'm at work right now but I'll test this
sometime this evening and (hopefully) close the ticket this evening. Thank
you Andy!

odellus, 2022-04-29:
I see this when I purge `moveit*` before doing `colcon build`
```
apt-get purge -y ros2-${ROS_DISTRO}-moveit*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros2-galactic-moveit*
E: Couldn't find any package by glob 'ros2-galactic-moveit*'
E: Couldn't find any package by regex 'ros2-galactic-moveit*'
```
and I'm still seeing the same error
```
/home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
/home/thomas/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:76: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:358: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit2_tutorials [38.1s, exited with code 2]

Summary: 43 packages finished [1min 29s]
  1 package failed: moveit2_tutorials
  2 packages had stderr output: moveit2_tutorials moveit_visual_tools
```

odellus, 2022-04-30:
I removed the `apt-get purge -y ros2-${ROS_DISTRO}-moveit*` line from the dockerfile because the error it was throwing when it could not find any ros2-galactic-moveit packages was preventing the colcon build.

Colcon is building moveit from source too.
```
Starting >>> moveit_ros
Starting >>> moveit_setup_assistant
Finished <<< pilz_industrial_motion_planner [2.38s]
Starting >>> moveit_planners
Finished <<< moveit_ros [0.54s]
Finished <<< moveit_planners [0.52s]
Starting >>> moveit
Starting >>> moveit_runtime
Finished <<< moveit_setup_assistant [0.97s]
Finished <<< moveit_runtime [0.58s]
Finished <<< moveit [0.61s]
```
so the problem **definitely** isn't that colcon is trying to link to dpkg install moveit libraries/headers in `/opt`

TomGrimwood, 2022-05-03:
I have been trying to install MoveIt2 for the last two days, and have reinstalled a fresh Ubuntu 20.04. Building MoveIt2 following https://moveit.ros.org/install-moveit2/source/ for ROS Galactic results in 
''''
/home/tom/ws_moveit2/src/moveit2/moveit_core/collision_detection/src/collision_matrix.cpp: In constructor ‘collision_detection::AllowedCollisionMatrix::AllowedCollisionMatrix(const srdf::Model&)’:
/home/tom/ws_moveit2/src/moveit2/moveit_core/collision_detection/src/collision_matrix.cpp:61:39: error: ‘const class srdf::Model’ has no member named ‘getNoDefaultCollisionLinks’
   61 |   for (const std::string& name : srdf.getNoDefaultCollisionLinks())

''''

and building following https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html
results in 

''''
/home/tom/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
/home/tom/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:63: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:380: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit2_tutorials [45.2s, exited with code 2]

Summary: 8 packages finished [1min 31s]
  1 package failed: moveit2_tutorials
  2 packages had stderr output: launch_param_builder moveit2_tutorials
''''

I am utterly confused as I am following all commands to exactly?

I am not familiar with docker, would that simplify things?

Thanks

k-macdonald, 2022-05-03:
I am seeing the same issue. I am not running in a docker environment, but am on Ubuntu 20.04, building MoveIt2 for ROS Galactic, following the QuickStart instructions listed above. Upon the colcon build step I am getting the following error with the moveit2_tutorials package:

```
--- stderr: moveit2_tutorials                             
$MOVEIT2_BASE/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function 'int main(int, char**)':
$MOVEIT2_BASE/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: 'using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >' {aka 'class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >'} has no member named 'future'
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:63: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:380: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make: *** [Makefile:141: all] Error 2 
```

I did try removing & purging as was suggested above, but am still caught on this error. Any help with this issue is appreciated. Thank you!

odellus, 2022-05-04:
I'm seeing the same error as @k-macdonald and have a Dockerfile to reproduce the error above. @TomGrimwood if you're interested in getting started with Docker you can find the instructions to install it on Ubuntu [here](https://docs.docker.com/engine/install/ubuntu/). Much easier than reinstalling the entire OS for a fresh slate.

henningkayser, 2022-05-04:
@odellus are you using the `main` branch of moveit2_tutorials? For Galactic, it should be the `[galactic](https://github.com/ros-planning/moveit2_tutorials/tree/galactic)` branch.

odellus, 2022-05-04:
Should I be building moveit2 from source with the galactic branch as well? Their [instructions](https://moveit.ros.org/install-moveit2/source/) for  building moveit2 indicate pulling from the main branch.

It seems vcs import is failing after seeing these notes about the detached head.
```
=== ./rviz_visual_tools (git) ===
Cloning into '.'...
Note: switching to '4.1.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 445b998 4.1.2
```
The Dockerfile I'm using is located [here](https://gist.github.com/odellus/3076c6df1dc9d5bfeec0d28a56f5a256). I've got echo statements before and after `vcs import` so it's clean vcs is what's returning the error code (though not to stdout strangely enough).

k-macdonald, 2022-05-04:
@henningkayser I was able to build by following the instructions at https://moveit.ros.org/install-moveit2/source/ up to the point of colcon build, then manually cd'ing into src/moveit2_tutorials and git switching branches. Since this was all automatically pulled down, I didn't think to check the available branches of the repos.  

Noting that this was again, all outside of docker, and is on the main branch of moveit2 and now the galactic branch of moveit2_tutorials. To summarize in code:

Following (https://moveit.ros.org/install-moveit2/source) instructions up to:
```
git clone https://github.com/ros-planning/moveit2.git -b main
for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
```

```
cd $COLCON_WS/src/moveit2_tutorials
git fetch
git switch galactic
cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
```

For what it's worth, I'm still getting a depreciation warning for setup.py in the launch_param_builder package:

```
--- stderr: launch_param_builder
~/.local/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_param_builder [3.04s]
Summary: 9 packages finished [3.28s]
  1 package had stderr output: launch_param_builder
```

But this one seems like it could be weirdness with my python setup and/or something that can be ignored. I haven't tried running anything yet, but it seems like this has fixed the build issues. 

Thank you for the help!!

TomGrimwood, 2022-05-04:
Hi, my issues with compiling following the MoveIt2 tutorials quickstart has been resolved by selecting the galactic branch as @henningkayser  suggested. The vcs import does complain as @odellus notes but this does not cause any issues.

odellus, 2022-05-05:
I took y'all's advice and built with `colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release` instead of `colcon build --mixin release` and it worked. Not sure how to keep the notes from git from causing my Dockerfile build from erroring out, but I'm able to run the "base" container and build moveit2_tutorials. Closing.

closed 2022-05-05 by odellus

1092technic, 2022-05-11:
I had the exact same issue. Tried installing Noetic instead, no luck either.

Get your sh*t together people!