moveit/moveit2 #1194: servo_teleop for a custom robot not working

kartiksoni01, 2022-04-23:
## Invalid move group name: `panda_arm` error while using servo_teleop.launch.py in moveit2
Hello, I am working on creating a package to teleop control a ur10 robotic arm using keyboard keys. The moveit2 comes with a launch file servo_teleop.launch.py that enables the panda_arm to be teleop controlled. I used the same launch files to make changes acc. to my robot and tried to implement the same, but when launched, it shows some unusual error. Below is my launch file:
 

```diff
import os
import yaml
from launch import LaunchDescription
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode
from launch.actions import ExecuteProcess
import xacro

def load_file(package_name, file_path):
    package_path = get_package_share_directory(package_name)
    absolute_file_path = os.path.join(package_path, file_path)

    try:
        with open(absolute_file_path, "r") as file:
            return file.read()
    except EnvironmentError:  # parent of IOError, OSError *and* WindowsError where available
        return None

def load_yaml(package_name, file_path):
    package_path = get_package_share_directory(package_name)
    absolute_file_path = os.path.join(package_path, file_path)

    try:
        with open(absolute_file_path, "r") as file:
            return yaml.safe_load(file)
    except EnvironmentError:  # parent of IOError, OSError *and* WindowsError where available
        return None

def generate_launch_description():
    # Get parameters for the Servo node
    servo_yaml = load_yaml("ur10_config", "config/simulated_config.yaml")
    servo_params = {"ur10_config": servo_yaml}

    # Get URDF and SRDF
    robot_description_config = xacro.process_file(
        os.path.join(
            get_package_share_directory("ur10_config"),
             "config", 
             "ur10.urdf.xacro",
        )
    )

    robot_description = {"robot_description": robot_description_config.toxml()}

    robot_description_semantic_config = load_file(
        "ur10_config", "config/ur10.srdf"
    )
    robot_description_semantic = {
        "robot_description_semantic": robot_description_semantic_config
    }

    # RViz
    rviz_config_file = (
        get_package_share_directory("ur10_config")
        + "/rviz/moveit.rviz"
    )
    rviz_node = Node(
        package="rviz2",
        executable="rviz2",
        name="rviz2",
        output="log",
        arguments=["-d", rviz_config_file],
        prefix=['gdb -ex run --args'],
        parameters=[robot_description, robot_description_semantic],
    )

    # ros2_control using FakeSystem as hardware
    ros2_controllers_path = os.path.join(
        get_package_share_directory("ur10_config"),
        "config",
        "ros_controllers.yaml",
    )
    ros2_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        prefix=['gdb -ex run --args'],
        parameters=[robot_description, ros2_controllers_path],
        output={
            "stdout": "screen",
            "stderr": "screen",
        },
    )

    # Load controllers
    load_controllers = []
    for controller in ["robot_controller", "joint_state_broadcaster"]:
        load_controllers += [
            ExecuteProcess(
                cmd=["ros2 run controller_manager spawner.py {}".format(controller)],
                shell=True,
                output="screen",
            )
        ]

    # Launch as much as possible in components
    container = ComposableNodeContainer(
        name="moveit_servo_demo_container",
        namespace="/",
        package="rclcpp_components",
        executable="component_container",
        prefix=['gdb -ex run --args'],
        composable_node_descriptions=[
            ComposableNode(
                package="robot_state_publisher",
                plugin="robot_state_publisher::RobotStatePublisher",
                name="robot_state_publisher",
                parameters=[robot_description],
            ),
            ComposableNode(
                package="tf2_ros",
                plugin="tf2_ros::StaticTransformBroadcasterNode",
                name="static_tf2_broadcaster",
                parameters=[{"child_frame_id": "/base_link", "frame_id": "/world"}],
            ),
            # ComposableNode(
            #     package="moveit_servo",
            #     plugin="moveit_servo::ServoNode",
            #     name="servo_server",
            #     prefix=['gdb -ex run --args'],
            #     parameters=[
            #         servo_params,
            #         robot_description,
            #         robot_description_semantic,
            #     ],
                
            #     # extra_arguments=[{"use_intra_process_comms": True}],
            # ),
            ComposableNode(
                package="moveit_servo",
                plugin="moveit_servo::JoyToServoPub",
                name="controller_to_servo_node",
                # extra_arguments=[{"use_intra_process_comms": True}],
            ),
            ComposableNode(
                package="joy",
                plugin="joy::Joy",
                name="joy_node",
                # extra_arguments=[{"use_intra_process_comms": True}],
            ),
        ],
        output="screen",
    )
    servo_node = Node(
        package="moveit_servo",
        executable="servo_node_main",
        prefix=['gdb -ex run --args'],
        parameters=[
            servo_params,
            robot_description,
            robot_description_semantic,
        ],
        output={
            "stdout": "screen",
            "stderr": "screen",
        },
    )
    return LaunchDescription(
        [rviz_node, ros2_control_node, servo_node, container] + load_controllers
    )

```
When launched the error shows up that says,
```
[servo_node_main-5] [New Thread 0x7fffe9b0a700 (LWP 39246)]
[servo_node_main-5] [ERROR] [1650702741.577403458] [moveit_robot_model.robot_model]: Group 'panda_arm' not found in model 'ur10'
[servo_node_main-5] [ERROR] [1650702741.577441880] [moveit_servo.servo_calcs]: Invalid move group name: `panda_arm`
[servo_node_main-5] terminate called after throwing an instance of 'std::runtime_error'
[servo_node_main-5]   what():  Invalid move group name
[servo_node_main-5] 
[servo_node_main-5] Thread 1 "servo_node_main" received signal SIGABRT, Aborted.
[servo_node_main-5] __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
[servo_node_main-5] 50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

I have checked and verified multiple times, i do not have panda_arm written on any section of the whole package, I wonder if any node callbacks internally have been constrained to the group_name 'panda_arm'. 
I am relatively a newbie in ROS2 space, a guidance would be appreciated.

AndyZe, 2022-04-23:
ROS2 seems to have issues if you have a debian installed but you are also building a package from source. Did you build moveit from source?

I would suggest removing any moveit_debians you have installed, then building moveit from source per directions [here](https://moveit.ros.org/install-moveit2/source/).

`sudo apt remove ros-$ROS_DISTRO-moveit-*`

Also make sure your package is sourced correctly. `source ./install/setup.bash` or similar

tylerjw, 2022-04-28:
My guess is that the 'panda_arm' group name is a parameter default in servo and that there is just something minor about how you are loading parameters for it.

kartiksoni01, 2022-04-28:
> ROS2 seems to have issues if you have a debian installed but you are also building a package from source. Did you build moveit from source?
>
> I would suggest removing any moveit_debians you have [...]

Done, made changes to my launch file (for servo) with reference to the servo_example.launch.py present in the moveit package. No success, the error mentioned above still seems to persist.

AkashKalghatgi, 2022-07-02:
Hello @kartiksoni01 
You may correct the following line inside your launch file:
`servo_params = {"ur10_config": servo_yaml}`
with:
`servo_params = {"moveit_servo": servo_yaml}`

KazumiLine, 2022-07-07:
Hello @AkashKalghatgi 
I also follow @kartiksoni01 's way, and change 
`servo_params = {"ur10_config": servo_yaml}`
to:
`servo_params = {"moveit_servo": servo_yaml}`

there's new error occurred
```
[component_container-3] [ERROR] [1657202349.957740900] [moveit_planning_scene.planning_scene]: Unknown frame: panda_link0
[rviz2-1] [INFO] [1657202352.409295200] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
```
but i can't find any used files containing  `panda_link0`

when i use servo_keyboard_input to control it, only tf moved.
[image]
how can i fix it, thx.

KazumiLine, 2022-07-08:
Hello @AndyZe , thanks for your answer.
Before i run the script, node list is empty.
```
root@DESKTOP-19D1MON:~# ros2 node list
root@DESKTOP-19D1MON:~# 
```

After i launched servo_teleop.launch.py and servo_keyboard_input:
```
root@DESKTOP-19D1MON:~# ros2 node list
/controller_manager
/controller_to_servo_node
/joint_state_broadcaster
/joy_node
/launch_ros_987
/moveit_servo_demo_container
/robot_controller
/robot_state_publisher
/rviz2
/rviz2_private
/servo_keyboard_input
/servo_server
/servo_server_private
/static_tf2_broadcaster
/transform_listener_impl_5648bccdbeb0
/transform_listener_impl_7f6574001ab0
root@DESKTOP-19D1MON:~#
```

i also find something might caused the problem, will the default value when file loading contains panda_link0?
```
[component_container-3] [INFO] [1657244792.268655800] [moveit_robot_model.robot_model]: Loading robot model 'ur10'...
[component_container-3] [INFO] [1657244792.268667500] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[component_container-3] [WARN] [1657244792.294114700] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
```

and when i running the keyboard input, it shows error:
```
[component_container-3] [WARN] [1657245383.721528100] [moveit_servo.servo_calcs]: Stale command. Try a larger 'incoming_command_timeout' parameter?
[component_container-3] [WARN] [1657245384.741565300] [moveit_servo.servo_calcs]: Ignoring joint shoulder_link
```

kalghatgi, 2022-07-08:
> Hello @AkashKalghatgi I also follow @kartiksoni01 's way, and change `servo_params = {"ur10_config": servo_yaml}` to: `servo_params = {"moveit_servo": servo_yaml}`
> 
> there's new error occurred
> 
> ```
> [component_container-3] [ERROR] [1657202349.957740900] [moveit_planning_scene.planning_scene]: Unknown frame: panda_link0
> [rviz2-1] [INFO] [1657202352.409295200] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
> ```
> 
> but i can't find any used files containing `panda_link0`
> 
> when i use servo_keyboard_input to control it, only tf moved. [image] how can i fix it, thx.
> 
> below is my launch file:
> 
> ```
> import os
> import yaml
> from launch import LaunchDescription
> from launch_ros.actions import Node
> from ament_index_python.packages import get_package_share_directory
> from launch_ros.actions import ComposableNodeContainer
> from launch_ros.descriptions import ComposableNode
> from launch.actions import ExecuteProcess
> import xacro
> from launch.substitutions import Command, FindExecutable, LaunchConfiguration, PathJoinSubstitution
> from launch_ros.substitutions import FindPackageShare
> 
> 
> def load_file(package_name, file_path):
>     package_path = get_package_share_directory(package_name)
>     absolute_file_path = os.path.join(package_path, file_path)
> 
>     try:
>         with open(absolute_file_path, "r") as file:
>             return file.read()
>     except EnvironmentError:  # parent of IOError, OSError *and* WindowsError where available
>         return None
> 
> 
> def load_yaml(package_name, file_path):
>     package_path = get_package_share_directory(package_name)
>     absolute_file_path = os.path.join(package_path, file_path)
> 
>     try:
>         with open(absolute_file_path, "r") as file:
>             return yaml.safe_load(file)
>     except EnvironmentError:  # parent of IOError, OSError *and* WindowsError where available
>         return None
> 
> 
> def generate_launch_description():
>     # Get parameters for the Servo node
>     servo_yaml = load_yaml("ur10_config", "config/simulated_config.yaml")
>     servo_params = {"moveit_servo": servo_yaml}
> 
>     # Get URDF and SRDF
>     robot_description_config = xacro.process_file(
>         os.path.join(
>             get_package_share_directory("ur10_config"),
>              "config", 
>              "ur10.urdf.xacro",
>         )
>     )
> 
>     robot_description = {"robot_description": robot_description_config.toxml()}
> 
>     robot_description_semantic_config = load_file(
>         "ur10_config", "config/ur10.srdf"
>     )
[... 71 lines cut ...]
>                 ],
>                 extra_arguments=[{"use_intra_process_comms": True}],
>             ),
>             ComposableNode(
>                 package="moveit_servo",
>                 plugin="moveit_servo::JoyToServoPub",
>                 name="controller_to_servo_node",
>                 extra_arguments=[{"use_intra_process_comms": True}],
>             ),
>             ComposableNode(
>                 package="joy",
>                 plugin="joy::Joy",
>                 name="joy_node",
>                 # extra_arguments=[{"use_intra_process_comms": True}],
>             ),
>         ],
>         output="screen",
>     )
>     return LaunchDescription(
>         [rviz_node, ros2_control_node, container] + load_controllers
>     )
> ```

In the moveit source file named "planning_scene.cpp", you may find a function "processCollisionObjectAdd(const moveit_msgs::msg::CollisionObject& object)" that checks the availability of frame name "panda_link0" within the group, which it isn't. Therefore, gives out the mentioned error. 
I see that inside your simulated_cnfig.yaml, you've specified the planning_frame and robot_link_command_frame to be "base_link", but somehow that's not being taken as an input. I think this is where the problem lies at.

KazumiLine, 2022-07-08:
Thank you for your enthusiastic responses.
I finally got it working.

It looks like that bug `Unknown frame: panda_link0` doesn't seem to affect its working.
Actually the error is because I set the initial position at the singularity, so it can't moved.

Now everything works fine, thank you guys.

kalghatgi, 2022-07-08:
> 

Well, how did you manage to get it out of singularity? did you make any changes inside the initial_positions.yaml from 0.0 to something else?

KazumiLine, 2022-07-08:
Yes, i changed initial_positions.yaml from 0.0 to
```
initial_positions:
  elbow_joint: 2.0
  shoulder_lift_joint: -1.57
  shoulder_pan_joint: 2.0
  wrist_1_joint: -1.57
  wrist_2_joint: 2.0
  wrist_3_joint: 2.0
```

and every thing works fine

[ur10_realtime_servo.zip](https://github.com/ros-planning/moveit2/files/9072613/ur10_realtime_servo.zip)

henningkayser, 2023-01-26:
Looks like this has been fixed a while ago, closing

closed 2023-01-26 by henningkayser

cboostpverwimp, 2024-09-17:
This error is still active in the "humble" branch.

> It looks like that bug `Unknown frame: panda_link0` doesn't seem to affect its working. Actually the error is because I set the initial position at the singularity, so it can't moved.

For those wondering, this is caused by: 
>``` python
>             ComposableNode(
>                package="moveit_servo",
>                plugin="moveit_servo::JoyToServoPub",
>                name="controller_to_servo_node",
>                extra_arguments=[{"use_intra_process_comms": True}],
>```
The JoyToServoPub plugin is based on the [joystick_servo_example.cpp](https://github.com/moveit/moveit2/blob/humble/moveit_ros/moveit_servo/src/teleop_demo/joystick_servo_example.cpp) which declares:
``` cpp
const std::string EEF_FRAME_ID = "panda_hand";
const std::string BASE_FRAME_ID = "panda_link0";
```
And later publishes a `/planning_scene` using `panda_link0` as `frame_id`. This also means the Xbox controller will not work when using another arm then the panda.
Replacing the plugin with a node or plugin that publishes a `/planning_scene` should fix this and some future errors.

atomoclast, 2024-12-06:
@cboostpverwimp Thanks for highlighting this is still an issue. I'm using the Debians  on ROS2 Humble right now. 

However, I'm attempting to use the servo initialization steps as detailed in this example: [Realtime Arm Servoing — MoveIt Documentation: Humble documentation](https://moveit.picknik.ai/humble/doc/examples/realtime_servo/realtime_servo_tutorial.html#initializing-servo), and I'm seeing the following error: 

```
[ERROR] [1733500940.924144413] [moveit_robot_model.robot_model]: Group 'panda_arm' not found in model 'custom_robot'
[ERROR] [1733500940.924153628] [moveit_servo.servo_calcs]: Invalid move group name: `panda_arm`
```

What's interesting is that I'm not trying to use the teleop demo as listed above, so not sure where this hardcoded issue is coming from?

I'm setting up the servo parameters by passing in my custom nodes. Do you know where this may be coming from?

atomoclast, 2024-12-10:
> [@cboostpverwimp](https://github.com/cboostpverwimp) Thanks for highlighting this is still an issue. I'm using the Debians on ROS2 Humble right now.
>
> However, I'm attempting to use the servo [...]

To reply back to this, my issue is that I didn't have a servo config yaml as shown in [this example for panda arms](https://github.com/moveit/moveit2/blob/humble/moveit_ros/moveit_servo/config/panda_simulated_config.yaml), and I wasn't passing it along properly as shown in this [launch file](https://github.com/moveit/moveit2_tutorials/blob/3e6576410e3322c86023ce9b35359fc73afc0de5/doc/examples/realtime_servo/launch/servo_cpp_interface_demo.launch.py#L37).

Once I added that, it seems to be starting now.