moveit/moveit2 #1205: MoveIt 2 Source Build - Linux - colcon build errors

TomGrimwood, 2022-05-02:
### Description
Attempting to build MoveIt2 with ROS Galactic installed from either binary or source, installed in the directory ~/ros2_galactic/ros2-linux/setup.bash 

### Your Environment
ROS Distro: Galactic
OS Version: Ubuntu 20.04
Source or Binary build: Binary

### Steps to reproduce

Follow instructions at https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Binary.html and https://moveit.ros.org/install-moveit2/source/ 

### Expected behaviour
The build should succeed

### Actual behaviour
The build does not succeed

### Backtrace or Console output

````
tom@tom-H87M-D3H:~/ws_moveit2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> srdfdom
Starting >>> geometric_shapes
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Finished <<< moveit_common [0.43s]
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_panda_description [0.99s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_fanuc_description [0.87s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.83s]
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_fanuc_moveit_config [1.14s]
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_resources_pr2_description [2.49s]
Starting >>> moveit_resources
Finished <<< moveit_resources_prbt_support [2.45s]
Finished <<< moveit_resources [2.90s]
Finished <<< srdfdom [19.2s]
Starting >>> moveit_configs_utils
Finished <<< moveit_configs_utils [2.46s]
[Processing: geometric_shapes]
[Processing: geometric_shapes]
--- stderr: geometric_shapes
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimpTargets.cmake:54 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimp-config.cmake:1 (include)
  CMakeLists.txt:41 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< geometric_shapes [1min 22s]
Starting >>> moveit_core
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
--- stderr: moveit_core
/home/tom/ws_moveit2/src/moveit2/moveit_core/collision_detection/src/collision_matrix.cpp: In constructor ‘collision_detection::AllowedCollisionMatrix::AllowedCollisionMatrix(const srdf::Model&)’:
/home/tom/ws_moveit2/src/moveit2/moveit_core/collision_detection/src/collision_matrix.cpp:61:39: error: ‘const class srdf::Model’ has no member named ‘getNoDefaultCollisionLinks’
   61 |   for (const std::string& name : srdf.getNoDefaultCollisionLinks())
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/tom/ws_moveit2/src/moveit2/moveit_core/collision_detection/src/collision_matrix.cpp:64:37: error: ‘const class srdf::Model’ has no member named ‘getEnabledCollisionPairs’; did you mean ‘getDisabledCollisionPairs’?
   64 |   for (auto const& collision : srdf.getEnabledCollisionPairs())
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                     getDisabledCollisionPairs
make[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/build.make:89: collision_detection/CMakeFiles/moveit_collision_detection.dir/src/collision_matrix.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2305: collision_detection/CMakeFiles/moveit_collision_detection.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [2min 40s, exited with code 2]

Summary: 11 packages finished [4min 3s]
  1 package failed: moveit_core
  2 packages had stderr output: geometric_shapes moveit_core
  29 packages not processed

````

henningkayser, 2022-05-04:
@TomGrimwood It looks like you are trying to compile the `main` branch on Galactic, which is not really supported anymore. The galactic branch does not include the changes that would produce this build error, so better try that one. If you still want to compile `main` on Galactic, you would at least have to pull in srdfdom as listed in the [repos](https://github.com/ros-planning/moveit2/blob/main/moveit2.repos) file. Please be aware, that `main` currently only supports Rolling+Jammy though.

TomGrimwood, 2022-05-07:
Could the instructions at https://moveit.ros.org/install-moveit2/source/ be changed for the galactic installation? it says to compile the main branch, which you say is incorrect for galactic.

JafarAbdi, 2022-05-09:
Fixed with https://github.com/ros-planning/moveit.ros.org/pull/702

closed 2022-05-09 by JafarAbdi