moveit/moveit2 #1209: Using different controllers for different joints in a robot

sampreets3, 2022-05-05:
### Issue

Hello, 

I am currently working on a project which uses a [Doosan A0509](https://github.com/doosan-robotics/doosan-robot2) collaborative robot arm on top of a custom mobile base. The mobile base is as of now a separate entity, controlled by sending signals through push-buttons. What I am able to do is add the mobile base as a prismatic joint in the srdf, and include the particular prismatic joint in the planning group. This enables me to be able to generate a trajectory which involves a certain amount of movement for the mobile base. 

What I found that I can do is I can connect the electrical signals necessary for the control of the mobile base to the Doosan robot control box, and use the services provided by doosan to set and reset outputs of the robot. This would enable me to control the mobile base with the Doosan robot. 

My point of struggle at the moment is trying to figure out how I can extract a part of the plan, retrieve the amount of distance I need to move the mobile base, and send the appropriate signals to the control box output. I can even work with a set of TF elements _(current TF of mobile base versus the desired TF of the mobile base)_ , but am currently not sure how I can extract that from the plan.

Any help on this issue would be highly appreciated, and thanks in advance!

Sampreet Sarkar

### System description

- MoveIt2 Foxy
- osrf/ros:foxy-desktop base image

henningkayser, 2022-06-02:
@sampreets3 Sorry for the late reply. I assume you are using MoveGroupInterface? The [plan()](https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html?highlight=mobile) function returns the robot trajectory with the [Plan](https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group_interface.h#L112) struct. You can parse the RobotTrajectory values, your base should be represented as linear joint offsets. No need to query TF imo, all you need to do is define the direction of the prismatic joint (positive X?) and read and process the values. If you really need some more advanced planning with the base, I'd recommend reading the corresponding tutorial on [mobile manipulation](https://moveit.picknik.ai/humble/doc/examples/mobile_base_arm/mobile_base_arm_tutorial.html).

sampreets3, 2022-06-03:
@henningkayser ,  thanks for the reply. I did take a look at the Plan struct and examined the object returned for a standard planning problem _(i.e., one in which I know everything will work and I know the displacement each link should make)_. I will follow your solution and simply extract the value needed to send to my custom controller. 

Thanks for the tutorial on Mobile manipulation. Just a quick question tho, is the tutorial compatible with MoveIt2 Foxy? I am working within a docker container, and I'd like to stay with Foxy as long as I can without explicitly needing to upgrade.

henningkayser, 2022-06-03:
> @henningkayser , thanks for the reply. I did take a look at the Plan struct and examined the object returned for a standard planning problem _(i.e., one in which I know everything will work and I [...]

Sounds good!

> Thanks for the tutorial on Mobile manipulation. Just a quick question tho, is the tutorial compatible with MoveIt2 Foxy? I am working within a docker container, and I'd like to stay with Foxy as long as I can without explicitly needing to upgrade.

The required joint model types made it into Foxy, so the tutorial should work in theory. Maybe @DLu knows this better.

DLu, 2022-06-09:
Unfortunately, we had to change the `control_msgs` API to get mobile manipulation to work, and [that change was only made in Galactic and later](https://github.com/ros-controls/control_msgs/pull/55)

closed 2023-06-30 by sampreets3