moveit/moveit2 #1217: Rolling Source Build Issues (FTC, so it may be just me...)

jeoseo, 2022-05-06:
### Description

Difficulties actually installing and properly running examples on Rolling, with many problems leading to open issues.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source or Binary build? Source, main branch

### Steps to reproduce

Created a fresh installation of Ubuntu 22.04, 

Follow the instructions from https://moveit.ros.org/install-moveit2/source/, except from Ubuntu 22.04 because Rolling no longer works on 20.04.

### Expected behaviour

I should be able to follow the instructions written to be able to build from source and be able to reproduce results from other issues, such as this one: https://github.com/ros-planning/moveit2/issues/1069.

### Actual behaviour

Building Moveit works fine, other than the deprecation warnings that appear here: https://github.com/ros-planning/moveit2/issues/1106. However, when trying to run the tutorials package, 
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: No module named 'moveit_configs_utils' 
```
The package is in the python path, and running ```ros2 pkg prefix moveit_configs_utils``` returns the package location correctly. For some reason, other python packages like launch_param_builder are successfully imported. For now, I used the hack fix below to get around this as I was not able to find a single difference in the way launch_param_builder and moveit_configs_utils were built and installed
```
import sys
sys.path.append('/home/jeoseo/ws_moveit2/src/moveit2/moveit_configs_utils')
```
Trying to run the launch file now gives the following issue, which is open here: https://github.com/ros2/launch/issues/588.
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): stdoutstderr is not a valid standard output config i.e. "screen", "log" or "both"  
```
I first tried using the workaround mentioned in the above issue, which is to build ROS2 from source using a specific commit of launch, until a fix is implemented. I ran into more issues building ROS2 from source (https://github.com/colcon/colcon-core/issues/506 is one of them, the other is that ament_lint also has a module import error), at which point I gave up and just removed the offending output config to be able to run even just demo code.  I have only had experience using ROS melodic/ other old versions before, so I would appreciate some guidance as to how I should be working with packages from ROS Rolling so I can contribute to more recent developement.

stephanie-eng, 2022-05-06:
Are you trying to run the Quickstart in Rviz `demo.launch.py` tutorial? This may be related to your launch issue: https://github.com/ros-planning/moveit2_tutorials/pull/357

jeoseo, 2022-05-06:
Yup, that's what I did for the 2nd error.

vatanaksoytezer, 2022-05-12:
@jeoseo are you still having similar issues? https://github.com/ros-planning/moveit2_tutorials/pull/357 has been merged, so I'm assuming it should be fine now, can you please confirm?

jeoseo, 2022-05-17:
I'm still having the first issue where moveit_configs_utils isn't imported without me adding the hack fix.

GradyM2M, 2022-06-06:
@jeoseo Hi, I have the same issue, any progress on this?

vatanaksoytezer, 2022-06-06:
@GradyM2M I don't think the error should exist anymore do you mind sharing your errors and environment?

GradyM2M, 2022-06-06:
my environment:
ROS Distro: Humble
OS Version: Ubuntu 22.04
Source, main branch
[image]

vatanaksoytezer, 2022-06-06:
@GradyM2M Seems like you have an earlier version of `srdfdom` or using a release version. Can you try removing srdfdom via `apt-get remove ros-humble-srdfdom` and source build the latest version of srdfdom using ros2 branch of https://github.com/ros-planning/srdfdom/tree/ros2 in your workspace?

sjahr, 2022-12-09:
@jeoseo I am closing this issue for now because I think it is outdated by now. Please re-open it or create a new one if you`re having still troubles

closed 2022-12-09 by sjahr