moveit/moveit2 #1233: Error when running "colcon build --mixin release"

1092technic, 2022-05-10:
### Description

Followed this tutorial on how to install moveit2[https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html]
ROS installed successfully. It all seems to work smoothly until I get to step "colcon build --mixin release" which fails at "moveit2_tutorials" 

### Your environment
* ROS Distro: Galactic
* OS Version: e.g. Ubuntu 20.04 VirtualBox
* Binary
* If binary, which release version? Galactic

### Steps to reproduce
From this tutorial: https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
#All required rosdeps installed successfully
cd ~/ws_moveit2
colcon build --mixin release

```
[0.334s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'moveit_resources_panda_moveit_config' is in: /opt/ros/galactic
	'moveit_resources_panda_description' is in: /opt/ros/galactic
	'launch_param_builder' is in: /opt/ros/galactic
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding launch_param_builder moveit_resources_panda_description moveit_resources_panda_moveit_config

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> moveit_resources_panda_description
Finished <<< moveit_resources_panda_description [0.12s]                  
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_fanuc_description [0.11s]                  
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.10s]                  
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_pr2_description [0.13s]                  
Starting >>> moveit_visual_tools
Finished <<< moveit_visual_tools [0.22s]                     
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_fanuc_moveit_config [0.11s]                
Starting >>> launch_param_builder
Finished <<< launch_param_builder [0.79s]          
Starting >>> moveit2_tutorials
--- stderr: moveit2_tutorials                             
/home/ozymandias/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp: In function ‘int main(int, char**)’:
/home/ozymandias/ws_moveit2/src/moveit2_tutorials/doc/examples/planning_scene_ros_api/src/planning_scene_ros_api_tutorial.cpp:164:77: error: ‘using SharedFuture = class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’ {aka ‘class std::shared_future<std::shared_ptr<moveit_msgs::srv::ApplyPlanningScene_Response_<std::allocator<void> > > >’} has no member named ‘future’
  164 |   response_future = planning_scene_diff_client->async_send_request(request).future.share();
      |                                                                             ^~~~~~
make[2]: *** [doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/build.make:63: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/src/planning_scene_ros_api_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:380: doc/examples/planning_scene_ros_api/CMakeFiles/planning_scene_ros_api_tutorial.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit2_tutorials [5.51s, exited with code 2]

Summary: 7 packages finished [7.34s]
  1 package failed: moveit2_tutorials
  1 package had stderr output: moveit2_tutorials
  1 package not processed
```
### Actual behaviour
Tell us what happens instead
Fails

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

1092technic, 2022-05-11:
@henningkayser Galactic

henningkayser, 2022-05-11:
> @henningkayser Galactic

I actually meant moveit2_tutorials, not the moveit2 repo. The particular line of code that fails doesn't exist in the `galactic` branch. The instructions to use the `galactic` branch have just been fixed recently.

closed 2022-06-02 by henningkayser

tsadarsh, 2022-06-11:
I have a similar issue that @1092technic faced. My `colcon build --mixin release` fails while building `moveit_visual_tools`. I have pasted the full output in [gist.github.com].(https://gist.githubusercontent.com/tsadarsh/8d22119ee2975f2e27635c2a725c0c00/raw/852b153945656943a074d2936980ed1610848734/colcon_build_fail.log)

Should I create a new issue regarding this?

TZECHIN6, 2022-12-30:
Hi guys, I have faced the same issue all you have mentioned in above. I am also in `galactic` branch, and just following the tutorial.. no luck so far

CoreJK, 2024-07-05:
I have faced the same issuse in **foxy** branch

1092technic, 2024-07-07:
@CoreJK I got through by a clean Ubuntu instance. If you tried to install the wrong version and then attempt to correct it, it will NOT work. You need to use a clean Ubuntu VM instance(I ran mine on virtual box)