moveit/moveit2 #1237: Cannot get moveit2 to work after following tutorial

1092technic, 2022-05-11:
### Description

Followed this tutorial (https://moveit.ros.org/install-moveit2/binary/) on how to install Moveit. I am using ROS Galactic, obtained by following this guide (https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html). When trying to build moveit2_tutorials following this guide(https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html) I get an error. 

BTW why so many guides? Noetic, Galactic, Rolling? You need like 5 different tutorials to get somewhere with this thing

Important detail: this is the 5th consecutive time I am attempting this. Last time I had a different error compiling moveit_tutorials, now something new came up.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 20.04 Focal
* Binary
* If binary, which release version? Movelt 2.0

### Steps to reproduce

From this tutorial: https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html

`cd ~/ws_moveit2/src
git clone -b galactic https://github.com/ros-planning/moveit2_tutorials.git
vcs import < moveit2_tutorials/moveit2_tutorials.repos`

### Expected behaviour
Should not get an error

### Actual behaviour
Fails

### Backtrace or Console output

`colcon build --mixin release
Starting >>> geometric_shapes
Finished <<< geometric_shapes [0.47s]                     
Starting >>> moveit_msgs
Finished <<< moveit_msgs [2.86s]                      
Starting >>> moveit_common
Finished <<< moveit_common [0.11s]                   
Starting >>> moveit_core
--- stderr: moveit_core                             
/home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/src/transforms.cpp: In member function ‘void moveit::core::Transforms::setTransform(const TransformStamped&)’:
/home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/src/transforms.cpp:144:58: error: no matching function for call to ‘transformToEigen(const TransformStamped&)’
  144 |     Eigen::Isometry3d t = tf2::transformToEigen(transform);
      |                                                          ^
In file included from /home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/src/transforms.cpp:42:
/opt/ros/noetic/include/tf2_eigen/tf2_eigen.h:48:20: note: candidate: ‘Eigen::Isometry3d tf2::transformToEigen(const Transform&)’
   48 |  Eigen::Isometry3d transformToEigen(const geometry_msgs::Transform& t) {
      |                    ^~~~~~~~~~~~~~~~
/opt/ros/noetic/include/tf2_eigen/tf2_eigen.h:48:69: note:   no known conversion for argument 1 from ‘const TransformStamped’ {aka ‘const geometry_msgs::msg::TransformStamped_<std::allocator<void> >’} to ‘const Transform&’ {aka ‘const geometry_msgs::Transform_<std::allocator<void> >&’}
   48 |  Eigen::Isometry3d transformToEigen(const geometry_msgs::Transform& t) {
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/opt/ros/noetic/include/tf2_eigen/tf2_eigen.h:58:19: note: candidate: ‘Eigen::Isometry3d tf2::transformToEigen(const TransformStamped&)’
   58 | Eigen::Isometry3d transformToEigen(const geometry_msgs::TransformStamped& t) {
      |                   ^~~~~~~~~~~~~~~~
/opt/ros/noetic/include/tf2_eigen/tf2_eigen.h:58:75: note:   no known conversion for argument 1 from ‘const TransformStamped’ {aka ‘const geometry_msgs::msg::TransformStamped_<std::allocator<void> >’} to ‘const TransformStamped&’ {aka ‘const geometry_msgs::TransformStamped_<std::allocator<void> >&’}
   58 | Eigen::Isometry3d transformToEigen(const geometry_msgs::TransformStamped& t) {
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/src/transforms.cpp: In member function ‘void moveit::core::Transforms::copyTransforms(std::vector<geometry_msgs::msg::TransformStamped_<std::allocator<void> > >&) const’:
/home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/src/transforms.cpp:166:53: error: no match for ‘operator=’ (operand types are ‘__gnu_cxx::__alloc_traits<std::allocator<geometry_msgs::msg::TransformStamped_<std::allocator<void> > >, geometry_msgs::msg::TransformStamped_<std::allocator<void> > >::value_type’ {aka ‘geometry_msgs::msg::TransformStamped_<std::allocator<void> >’} and ‘geometry_msgs::TransformStamped’ {aka ‘geometry_msgs::TransformStamped_<std::allocator<void> >’})
  166 |     transforms[i] = tf2::eigenToTransform(it->second);
      |                                                     ^
In file included from /opt/ros/galactic/include/geometry_msgs/msg/transform_stamped.hpp:7,
                 from /home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/include/moveit/transforms/transforms.h:39,
                 from /home/ozymandias/ws_moveit2/src/moveit2/moveit_core/transforms/src/transforms.cpp:37:
/opt/ros/galactic/include/geometry_msgs/msg/detail/transform_stamped__struct.hpp:37:8: note: candidate: ‘geometry_msgs::msg::TransformStamped_<std::allocator<void> >& geometry_msgs::msg::TransformStamped_<std::allocator<void> >::operator=(const geometry_msgs::msg::TransformStamped_<std::allocator<void> >&)’
   37 | struct TransformStamped_
      |        ^~~~~~~~~~~~~~~~~
/opt/ros/galactic/include/geometry_msgs/msg/detail/transform_stamped__struct.hpp:37:8: note:   no known conversion for argument 1 from ‘geometry_msgs::TransformStamped’ {aka ‘geometry_msgs::TransformStamped_<std::allocator<void> >’} to ‘const geometry_msgs::msg::TransformStamped_<std::allocator<void> >&’
/opt/ros/galactic/include/geometry_msgs/msg/detail/transform_stamped__struct.hpp:37:8: note: candidate: ‘geometry_msgs::msg::TransformStamped_<std::allocator<void> >& geometry_msgs::msg::TransformStamped_<std::allocator<void> >::operator=(geometry_msgs::msg::TransformStamped_<std::allocator<void> >&&)’
/opt/ros/galactic/include/geometry_msgs/msg/detail/transform_stamped__struct.hpp:37:8: note:   no known conversion for argument 1 from ‘geometry_msgs::TransformStamped’ {aka ‘geometry_msgs::TransformStamped_<std::allocator<void> >’} to ‘geometry_msgs::msg::TransformStamped_<std::allocator<void> >&&’
make[2]: *** [transforms/CMakeFiles/moveit_transforms.dir/build.make:63: transforms/CMakeFiles/moveit_transforms.dir/src/transforms.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2236: transforms/CMakeFiles/moveit_transforms.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [4.58s, exited with code 2]

Summary: 3 packages finished [8.29s]
  1 package failed: moveit_core
  1 package had stderr output: moveit_core
  33 packages not processed`

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

JafarAbdi, 2022-05-11:
You're mixing different ROS distributions (noetic & galactic), you need to build with a clean shell (none of the env variables mention noetic), do you have a line in your bashrc that source noetic's specific `setup.bash` file?

> BTW why so many guides? Noetic, Galactic, Rolling? You need like 5 different tutorials to get somewhere with this thing

You only need to follow one of them which correspond to the distribution you're using

1092technic, 2022-05-12:
@JafarAbdi I did. 
You didn't even read the issue: I indicated I attempted Galactic, NOT Noetic. 

I tried ALL THREE distros consecutively to see if something worked, and removed them with command "sudo apt remove ~nros-galactic-* && sudo apt autoremove" modified specific to each distro. After sourcing, I checked for the correct distro with "echo $ROS_DISTRO" and it was right. It all worked fine until I try to build colcon: colcon build --mixin release

The amount of people having trouble, some exactly similar to mine from my previous attempt, is a testament that something is very wrong with the way the guides are made

JafarAbdi, 2022-05-12:
If you read the error message carefully, you will see that it included a header file from noetic and another one from galactic, which is causing the issue for you

/opt/ros/**noetic**/include/tf2_eigen/tf2_eigen.h
/opt/ros/**galactic**/include/geometry_msgs/msg/detail/transform_stamped__struct.hpp

closed 2022-05-14 by JafarAbdi