moveit/moveit2 #1241: Build Failed<< moveit_core -- libmoveit_butterworth_filter.so.2.3.2: undefined reference

AndresRoc, 2022-05-12:
### Description
I am building moveit2 from source following this [instructions ](https://moveit.picknik.ai/galactic/doc/tutorials/getting_started/getting_started.html) and I have an error with moveit_core.

### Your environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? main

### Steps to reproduce
 The error presents while executing:
`colcon build --mixin release`

### Expected behaviour
No errors

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

andres@ubuntu:~/ros2_ws$ colcon build --mixin release
Starting >>> geometric_shapes
Finished <<< geometric_shapes [0.70s]                     
Starting >>> moveit_msgs
Finished <<< moveit_msgs [4.59s]                      
Starting >>> moveit_common
Finished <<< moveit_common [0.43s]                   
Starting >>> moveit_core
--- stderr: moveit_core                              
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::getCurrentlyLoadingLibraryName[abi:cxx11]()'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::getPluginBaseToFactoryMapMapMutex()'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::hasANonPurePluginLibraryBeenOpened(bool)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::getCurrentlyActiveClassLoader()'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::AbstractMetaObjectBase::setAssociatedLibraryPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::getFactoryMapForBaseClass(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::AbstractMetaObjectBase::AbstractMetaObjectBase(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libmoveit_butterworth_filter.so.2.3.2: undefined reference to `class_loader::impl::AbstractMetaObjectBase::addOwningClassLoader(class_loader::ClassLoader*)'
collect2: error: ld returned 1 exit status
make[2]: *** [online_signal_smoothing/CMakeFiles/test_butterworth_filter.dir/build.make:210: online_signal_smoothing/test_butterworth_filter] Error 1
make[1]: *** [CMakeFiles/Makefile2:1376: online_signal_smoothing/CMakeFiles/test_butterworth_filter.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_core [2.32s, exited with code 2]

Summary: 3 packages finished [8.41s]
  1 package failed: moveit_core
  1 package had stderr output: moveit_core
  33 packages not processed

vatanaksoytezer, 2022-05-12:
Hello @AndresRoc thank you for the issue. Can you confirm you are using foxy branches of moveit2_tutorials and moveit2 (assuming you are using foxy)?

AndresRoc, 2022-05-12:
Hello @vatanaksoytezer here is a print of the branch and the environment:

andres@ubuntu:~/ros2_ws/src/moveit2$ git branch
* (HEAD detached at 2.3.2)
  main
andres@ubuntu:~/ros2_ws/src/moveit2$ printenv | grep -i ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
PWD=/home/andres/ros2_ws/src/moveit2
AMENT_PREFIX_PATH=/opt/ros/foxy
PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
LD_LIBRARY_PATH=/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/foxy/bin:/home/andres/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
ROS_DISTRO=foxy
OLDPWD=/home/andres/ros2_ws/src

vatanaksoytezer, 2022-05-12:
@AndresRoc please switch to foxy branch of moveit2 and moveit2_tutorials (via `git checkout` or `git switch` if you are using Foxy as your ROS distro. Let us know if you have the same issues.

AndresRoc, 2022-05-12:
Hello @vatanaksoytezer thanks for the quick response, I've checked out the new branches and compiled them again, I'll be back with an update soon!!

AndresRoc, 2022-05-12:
It seems like I have to repeat the same issue with every package. I found several that were at "Galactic's" branch. So far moveit_core has compiled correctly. There are 9 more packages to build. But I think so far the main issue has been resolved.
 Thanks a ton @vatanaksoytezer

vatanaksoytezer, 2022-05-12:
No worries, happy to help. Closing this issue for now, feel free to re-open it if you have any issues.

AndresRoc, 2022-05-12:
Final update, all colcon finished successfully!!! Thanks a ton!!

virajawate, 2023-02-15:
Hey, I am having a similar issue for building RRTstar global planner with ompl library. How can i solve this for noetic ROS?