moveit/moveit2 #1259: Help wanted: How to build the feature/msa branch?

marc-wittwer, 2022-05-19:
I'm really used to installing official releases. However, since the MSA for moveit2 is not yet merged into a release I would like to build it from source.

Is  this the right way of installing the feature branch (feature/msa) and launching MSA ?

I have ROS 2 Galactic installed.
```
// install moveit2 from the source guide: https://moveit.ros.org/install-moveit2/source/

// here the galactic branch gets checked out. Is this the place to swap galactic with feature/msa
git clone https://github.com/ros-planning/moveit2.git -b galactic

for repo in moveit2/moveit2.repos $(f="moveit2/moveit2_$ROS_DISTRO.repos"; test -r $f && echo $f); do vcs import < "$repo"; done
rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y

// Afterwards build in the workspace
cd $COLCON_WS
colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
```

Afterwards how would I launch the MSA?
```
// This is the old ROS1 way
roslaunch moveit_setup_assistant setup_assistant.launch
```

Thanks in advance for helping me out.

vatanaksoytezer, 2022-05-19:
Hello @marc-wittwer thank you for the issue. feature/msa branch is on Rolling (like the main branch, which is on Ubuntu 22.04) and scheduled to be merged before Humble release. Checking out feature/msa branch and following source build instructions for Rolling should suffice. Unfortunately main and rolling are pretty far from each other right now and it should not compile on galactic. If you don't want to switch your local installation I recommend using our rolling-source docker image, which you can find at https://hub.docker.com/r/moveit/moveit2. We've some upcoming documentation on how to use our docker images at https://github.com/ros-planning/moveit2_tutorials/pull/291. I would recommend just using the docker image, checking out feature branch in moveit2 and building it.

There is also some documentation on the architecture and quick start instructions in https://github.com/ros-planning/moveit2/blob/feature/msa/moveit_setup_assistant/README.md. Just to give you a quick estimate of MSA schedule: we are planning to merge it into main within a week and write more tutorials and documentation on the following three weeks, mostly at [World MoveIt Day](https://moveit.ros.org/events/world-moveit-day-2022/) in June 2 (Feel free to join us!).

Please let me know if you have any more questions or issues.

marc-wittwer, 2022-05-19:
@vatanaksoytezer 
Thanks for your reply.
I will use the docker approach.

I just started getting into ROS and moveit last month and I want to get to know the whole workflow.

My plan:
- Create simple multilink robot arm in urdf
- Create moveit config files with MSA (where I'm currently at)
- Launch Gazebo or RViz and control joints with a simple python script using the moveit commands via ROS

If I understood correctly, it is the recommended way to generate the moveit config files using the MSA. Writing these config files by hand does not make sense, right?

The World MoveIt Day sounds awesome. However, due to my limited understanding of the moveit library I won't be of too much help other than improving the documentation.

marc-wittwer, 2022-05-19:
@vatanaksoytezer 
I was successful in running the docker container and checking out the feature/msa branch. Afterwards I ran `colocon build` and `source install/setup.bash`

After executing `ros2 run moveit_setup_assistant moveit_setup_assistant` the GUI stated on my screen.
After selecting a new URDF file the GUI froze and showed the following alert window.

[image]

marc-wittwer, 2022-05-20:
It was basically just a cylinder.

```
<?xml version="1.0"?>
<robot name="myfirst">
  <link name="base_link">
    <visual>
      <geometry>
        <cylinder length="0.6" radius="0.2"/>
      </geometry>
    </visual>
  </link>
</robot>

```

marc-wittwer, 2022-05-20:
Edit: The panda.urdf and the fanuc.urdf seem to work. :+1:

closed 2022-06-09 by marc-wittwer