moveit/moveit2 #1261: [MSA] Port `moveit_setup_controllers`

DLu, 2022-05-19:
The `moveit_setup_controllers` package is currently ignored, but does compile. However, it still needs to be ported to generate [ros2_control](https://github.com/ros-controls/ros2_control) configurations.

In ROS 1, this step would generate two files:
 * `config/ros_controllers.yaml`
 * `launch/ros_controllers.launch`

The core information that is gathered is a list of controllers (specified by a name, type and list of joints).

My understanding of how it should be ported is based on the configurations in [`moveit_resources`](https://github.com/ros-planning/moveit_resources/). I believe the ROS 2 version should generate:
 * `config/moveit_controllers.yaml`
 * `config/ros2_controllers.yaml`
 * `config/initial_positions.yaml`
 * `config/[ROBOT_NAME].urdf.xacro`
 * `config/[ROBOT_NAME].ros2_control.xacro`

It seems relatively straightforward to take the list of controllers from the current setup step and write them info `moveit_controllers.yaml`. However, figuring out what information populates the rest requires a level of expertise I don't have.

I believe the process of setting up your robot for `ros2_control` could be its own wizard entirely, using this framework and different plugins.

AndyZe, 2022-05-26:
I'm not saying you have to do this by any means but if you can grab the information to fill out `moveit_controllers.yaml`, I think you could choose reasonable defaults for everything else:

- `ros2_controllers.yaml`: Requires the same joint information. You could have a `trajectory_controller` and a `joint_state_broadcaster` by default. I guess an `update_rate` of 100 is a good default.

- `initial_positions.yaml`: Requires the same list of joints. You could default to initial positions of zero -- the user can change it later if they need to.

- `config/[ROBOT_NAME].urdf.xacro`: OK, this one will be tough. Maybe can't use a wizard for it.

- `config/[ROBOT_NAME].ros2_control.xacro`: Requires the same list of joints. I think you could default to:

  <command_interface name="position"/>
  <state_interface name="position"/>
  
  This bit would be the same for every robot:
  
```
              <hardware>
                <plugin>fake_components/GenericSystem</plugin>
            </hardware>
```

JafarAbdi, 2022-05-27:
I think we could get rid of two files:

1- `initial_positions.yaml`: I add this to prevent having duplicate `*.ros2_control.xacro` file since we needed different start state for some tests, we could set the default values to 0.0 and let user change them if they need

2- `config/[ROBOT_NAME].urdf.xacro`: this was a lazy solution to add the `ros2_control` tag without modifying the original urdf file, we could have the moveit_config_utils update the urdf with that tag in python, I think I have a good solution, let me know if you want me to push it

AndyZe, 2022-05-27:
> I think we could get rid of two files:
> 
> 1- `initial_positions.yaml`: I add this to prevent having duplicate `*.ros2_control.xacro` file since we needed different start state for some tests, we could set the default values to 0.0 and let user change them if they need

Good idea!

> 2- `config/[ROBOT_NAME].urdf.xacro`: this was a lazy solution to add the `ros2_control` tag without modifying the original urdf file, we could have the moveit_config_utils update the urdf with that tag in python, I think I have a good solution, let me know if you want me to push it

Meh, I kind of like the modular setup we have now.

AndyZe, 2022-05-27:
> I think we could get rid of two files:
> 
> 1- `initial_positions.yaml`: I add this to prevent having duplicate `*.ros2_control.xacro` file since we needed different start state for some tests, we could set the default values to 0.0 and let user change them if they need

Thinking about this more, there is one use case for this. What if you want to define different initial positions for different scenarios? We actually do that here in `moveit_servo`:  https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/test/config/collision_start_positions.yaml

So I think we should keep the `initial_positions.yaml`

JafarAbdi, 2022-05-27:
@AndyZe I went ahead and implemented what I had in mind :D, please take a look at https://github.com/ros-planning/moveit2/pull/1283, does it make sense to you?

DLu, 2022-06-14:
Closed with #1299

closed 2022-06-14 by DLu