moveit/moveit2 #1269: Humble Release

henningkayser, 2022-05-24:
Tracking issue for the Humble Release steps:

- [ ] Generate Docker images
- [ ] Enable Humble CI jobs
- [ ] Tag Release version
- [ ] Bloom

vatanaksoytezer, 2022-05-26:
Released!: https://github.com/ros/rosdistro/pull/33378

closed 2022-05-26 by vatanaksoytezer

bernharl, 2022-06-02:
While it's now been released, I still cannot install it using apt. Is there anywhere I can check on the progress of getting it in the ROS2 humble repos?

gavanderhoorn, 2022-06-02:
[status_page/ros_humble_default.html?q=moveit](https://repo.ros2.org/status_page/ros_humble_default.html?q=moveit).

If/when the last square in the right column is green, you can use `apt` to install it from the main repositories.

You could already do it if you'd setup the testing repository though.

vatanaksoytezer, 2022-06-02:
We are just waiting a new sync from OR, and you'll be able to use in the main repos. You should be able to apt install if you include testing repositories as in https://github.com/ros-planning/moveit2/blob/main/.docker/ci-testing/Dockerfile#L9