moveit/moveit2 #1276: c++: fatal error: Killed signal terminated program cc1plus

dshMechMind, 2022-05-26:
### Description
I followed this instruction
https://moveit.ros.org/install-moveit2/source/
I could make sure very step be correct until the last build step:
`colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`
I got Error report every time `c++: fatal error: Killed signal terminated program cc1plus`
I tried `colcon build --event-handlers console_direct+ --cmake-args -DCMAKE_BUILD_TYPE=Debug --executor sequential`
and still received the same Error
### Your environment
* ROS Distro: foxy
* OS Version: Ubuntu 20.04
* Source or Binary build? source build
* If binary, which release version?
* If source, which branch? foxy

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output
[gist.github.com](https://gist.github.com/dshMechMind/126b2ba4eee0fcfe46d7b65598148f21)

JafarAbdi, 2022-05-26:
I think your PC run out of RAM, see https://github.com/ros-planning/moveit2/issues/538#issuecomment-874050488

closed 2022-05-26 by JafarAbdi

shaohedu, 2022-05-29:
Thanks a lot, I added 8g swap partition, it works!

tarinip26, 2023-02-18:
how did you do that? could you please help me out, i'm getting the same error on my parallel ubuntu 20.04 desktop version

dshMechMind, 2023-04-10:
My problem is because of out of memeory when compiling, you can monitor your systom memeory.
I solve this problem by add 8g swap partition, following this:
[add swap partition](https://askubuntu.com/questions/33697/how-do-i-add-swap-after-system-installation)