moveit/moveit2 #1279: GSoC: Create Python bindings for MoveIt 2

peterdavidfagan, 2022-05-26:
Hi there, 

I am currently working on creating Python bindings for MoveIt 2 as part of my [GSoC project](https://summerofcode.withgoogle.com/programs/2022/projects/AICLMf2p).

The goals of my project are to:

1. Redesign and implement a Python API for MoveIt 2 based on the original `moveit_commander` package
2. Create a tutorial demonstrating the usage of this API for learning based approaches (deep reinforcement learning)
3. Implement feature requests the come from the MoveIt community

I will be adding further details to this issue page and my project page overtime. Feel free to comment below or link feature requests that you feel may be relevant to this project. 

Have a great rest of the day.

peterdavidfagan, 2022-06-10:
Hi @tlpss, 

Thank you for these points. I have taken note of these and will look to incorporate the outlined functionalities into the Python API that gets created.  It would be great to have you test the initial library. 

Currently we are deciding what to bind and how the API should be structured. At the moment it looks like the functionality of the `moveit_cpp` library will be binded. The overall API design is still not finalized. I hope to provide demos using Isaac Sim.

tlpss, 2022-06-13:
> It would be great to have you test the initial library that results from this project.

Feel free to ping me, I would be interested to take a look. 

> Currently we are deciding what to bind and how the API should be structured. At the moment it looks like the functionality of the moveit_cpp library will be binded. The overall API design is still not finalized. I hope to provide demos using Isaac Sim so maybe this will be helpful for your third point.

Sounds great,in addition to ISAAC sim, Unity might also be an interesting choice as it is completely open source and does not require a beefy GPU (ISAAC sim requires an RTX GPU if I'm not mistaken). 

Best of luck with the project!

peterdavidfagan, 2022-06-29:
Hi everyone, 

The development of a library that is called `moveit_py` has begun; initially this library will consist of binding functionalities of `moveit_cpp` and creating useful python interfaces using these functionalities. As `moveit_cpp` continues to develop this library will develop alongside it. For those who are interested in tracking the development work for this library the [moveit_py](https://github.com/peterdavidfagan/moveit2/tree/moveit_py) branch on my fork should remain up to date.

I also have another repository [moveit_py_example](https://github.com/peterdavidfagan/moveit_py_example
) which will demonstrate examples of using the library. There's only one example there for the moment but I will be adding many more in the coming weeks (including isaac-sim examples which I have from work undertaken while onboarding). 

Finally I have started a project to track development work which can be found [here](https://github.com/users/peterdavidfagan/projects/4). I'll be adding tasks to this soon. 

Other than this update, I may share some of the interfaces in this thread to get your feedback on these so expect further updates in the coming weeks.

marc-wittwer, 2022-07-01:
Awesome. @peterdavidfagan   I will follow your progress :raised_hands: 

The python interface will lower the entry barrier for our research group as well. 

We will mainly use:
- Setting pose goals
- Setting collision objects into the planning scene
- Setting orientation constraints of the end effector while following a trajectory (e.g. moving an cup with liquid inside)

Best of luck with the project also from my side!

peterdavidfagan, 2022-07-01:
Thanks @marc-wittwer, I am really looking forward to the first release of the library. I'll be sure to take into account the use cases you mentioned above. It is great to have so much interest in the library already thank you for sharing the above.

larrylu0426, 2022-07-21:
Hi everyone, I'm working on a project about Panda's arm, and it depends on Python3 and ROS2/Moveit2/Ubuntu20.04. So I refactor the planning interface module with pybind11 on the foxy branch. Here's my repo link: [https://github.com/larrylu0426/moveit2-python](url). I keep all C++ interfaces and adapt the original Python interfaces. I have applied it in my project. So you can use it and give me many reviews :). 

MoveIt Command is ready to refactor.

peterdavidfagan, 2022-07-22:
Hi @larrylu0426, 

The above looks great, it sounds like you are interested in the same topics as this GSoC project. 

You can view the current status of the development [here](https://github.com/peterdavidfagan/moveit2/tree/moveit_py_dev/python/moveit_py) and a basic example of its usage [here](https://github.com/peterdavidfagan/moveit_py_example). I am starting to work on bindings for the planning_scene_interface as part of the package linked above. The bindings you created with pybind11 are likely similar to the ones I am developing, except the bindings I am creating lie in a separate package. 

If you are up for it we could hop on a call and discuss these bindings? Maybe some of your code can be integrated into the package I am creating as part of GSoC. 

I plan to have a tutorial where the planning_scene_interface is used by early next week. Once this is ready we could meet to discuss further.

larrylu0426, 2022-08-01:
Hi everyone, I have done the refactoring of the whole planning interface module and most parts of the moveit_commander module on main branch. Also, I have done a basic test by myself. Pls feel free to try it in your project :). The rest of remained code in moveit_commander will not be refactored from my side. I'm expecting the Peter's work. I think it will provide more powerful things for Python developers. :)

peterdavidfagan, 2022-08-01:
Hi @larrylu0426, 

The above refactoring is great and I think it will be useful for users of the planning_interface and moveit_commander. 

It would also be great to get your continued involvement in moveit_py, I'd be happy to schedule a call to discuss further. Also if you don't mind I would like to ask if you would be interested in testing the initial release I make of moveit_py.

larrylu0426, 2022-08-02:
> Hi @larrylu0426,
>
>
>
> The above refactoring is great and I think it will be useful for users of the planning_interface and moveit_commander.
>
>
>
> It would also be great to get your continued [...]

Great! I will try to use your released version for my work and report the usage result. We can keep in touch by email due to the time difference. 😄

peterdavidfagan, 2022-08-05:
Hi everyone, 

Thanks for being patient on the first release of `moveit_py`. I wished to provide a brief update on the current status and some exciting functionalities that become possible with the use of Python.

- The main functionalities of the C++ library `moveit_cpp` are now binded
- Many of the functionalities of RobotState and RobotModel are now binded
- Interacting with the planning scene via the planning scene monitor is now supported
- Support for auto-completion (we have added stub files to the project, they still need to be completed)
- The option to use Jupyter Notebooks to interactively write and execute code is being tested

I hope to push a broader set of changes this week related to the above points and start writing tutorials for the currently available functionality. I will send a notification here when all these changes have been pushed. I'll also try to provide more frequent updates from now until the official release of the library (which will likely be another while since documentation and tests need to be written while the library also needs to mature sufficiently and achieve a standard that can be approved and merged by my mentors). 

In the future, I also hope to add support for:
- APIs for interacting with a Servo Node
- Examples/support for reinforcement learning applications 

Hopefully this brief update gives you an idea of the current status. 

Have a great rest of the day and hopefully more updates to come soon. 

[image]

lorepieri8, 2022-09-08:
@peterdavidfagan When do you think the bindings will be releases? Happy to test them and share feedback!

AndyZe, 2022-09-16:
Nice. No, `GripperCommand` isn't used.

There's a service call to allow motion to drift in certain dimensions, but I don't think it's important enough to port to Python:

`test/servo_launch_test_common.hpp:    client_change_drift_dims_ = node_->create_client<moveit_msgs::srv::ChangeDriftDimensions>(`

Nothing else comes to mind

peterdavidfagan, 2022-09-16:
I would be happy to work towards adding support for the above functionality with your guidance. It would be helpful for enabling learning-based approaches. I also feel this is something other users would find valuable.

ronniethehood, 2022-10-27:
Hi @peterdavidfagan, this looks great. Any update on when it might be released?

peterdavidfagan, 2022-10-27:
> Hi @peterdavidfagan, this looks great. Any update on when it might be released?

Thanks @ronniethehood, apologies for the delay in the release of the library. The initial plan was to release at the end of this month, this is now more likely to be in mid-November. For context I am working on unit testing, tutorials and cleaning up the code. There will also be reviews from some of the maintainer team which should lead to additional improvements to the python library code. 

Once the initial release is made there will be continued support for feature requests and development.

vuva, 2022-11-24:
> > Hi @peterdavidfagan, this looks great. Any update on when it might be released?
> 
> Thanks @ronniethehood, apologies for the delay in the release of the library. The initial plan was to release at the end of this month, this is now more likely to be in mid-November. For context I am working on unit testing, tutorials and cleaning up the code. There will also be reviews from some of the maintainer team which should lead to additional improvements to the python library code.
> 
> Once the initial release is made there will be continued support for feature requests and development so if you start using the library after we release it and have a feature that you wish to be supported I am happy to work on this.

Hi @peterdavidfagan , is time parameterization supported in the initial version of the python binding? And sorry to ask this again, when will you release it?
Thank you

peterdavidfagan, 2022-11-26:
Hi @vuva, 

Apologies that you have had to ask this. Time parameterization will eventually be supported, it is not in the current set of code for the library but will be added. 

The initial plan was to release the library by the end of GSoC, unfortunately, this deadline was not met. I have set aside time in December and January just for the completion of this library so I hope to have code ready to be merged by the new year.

metanav, 2023-01-02:
Any ETA for the release?

peterdavidfagan, 2023-01-02:
Hi @metanav, 

The current [pr](https://github.com/ros-planning/moveit2/pull/1546) has been rebased to include the latest set of commits on the main branch of the MoveIt repository so the library code should be usable with the latest changes. 

I am finishing writing the tutorials and ensuring the API documentation is accurate this week. I may also make some API changes this week but otherwise, I would hope that this could be reviewed and merged soon. I expect to have my code ready for review by the end of this week but I am unsure how long it will take to be reviewed and merged thereafter. 

Also feature requests are welcome once the code has been merged.

sjahr, 2023-06-22:
@peterdavidfagan I guess this is done :tada: right?

peterdavidfagan, 2023-06-22:
Hi @sjahr, 

Apologies for delayed responses on my side recently. There are more features and improvements I wish to add to the library but I think this task can be closed for now. I will follow up with separate items for anything else I work on.

closed 2023-06-22 by peterdavidfagan