moveit/moveit2 #1289: WMD Track: DevOps

vatanaksoytezer, 2022-05-31:


JafarAbdi, 2022-06-01:
- Port moveit_task_constructor_core integration tests, requires porting:
  - tams_ur5_setup_moveit_config
  - pr2_moveit_config
  - pa10_shadow_moveit 
  - https://github.com/ros-planning/moveit_task_constructor/blob/ros2/core/test/CMakeLists.txt  
  - https://github.com/ros-planning/moveit_task_constructor/blob/ros2/core/package.xml

JafarAbdi, 2022-06-01:
- Port moveit_task_constructor_demo integration test https://github.com/ros-planning/moveit_task_constructor/blob/ros2/demo/CMakeLists.txt#L64-L65
  - Example of a ported integration test https://github.com/ros-planning/moveit_task_constructor/blob/ros2/core/test/move_to.launch.py

JafarAbdi, 2022-06-01:
Not ported tests for core features:
- https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/subframes_test.test
- https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/move_group_interface_cpp_test.test
- https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning/trajectory_execution_manager/test/test_execution_manager.test
- https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/moveit_cpp_test.test
- https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/move_group_pick_place_test.test
- https://github.com/ros-planning/moveit2/blob/main/moveit_ros/planning_interface/test/move_group_interface_cpp_test.test

mikeferguson, 2022-06-27:
@vatanaksoytezer should we rename this ticket? Looks like most of these things are not ticketed elsewhere (unlike our other WMD tickets)

vatanaksoytezer, 2022-06-28:
@mikeferguson yeah for sure, feel free to edit yourself or suggest any better names, I'm happy to change.