moveit/moveit2 #1292: WMD Track: MoveIt Config Utils

JafarAbdi, 2022-05-31:


DLu, 2022-05-31:
https://github.com/ros-planning/moveit2/issues/1297

JafarAbdi, 2022-06-01:
Refactor [MoveItConfigs](https://github.com/ros-planning/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py#L86-L213) to be decorated with dataclass

```py
from dataclasses import dataclass

@dataclass(slots=True)
class MoveItConfigs:
    something: Type = DefaultValue
    ...
```

This would save us from having the following for each parameter

```py
__slots__ = [
    "__something",
    ...
]

@property
def something(self):
    return self.__something

@something.setter
def something(self, value):
    self.__something = value
```

closed 2023-01-25 by henningkayser