moveit/moveit2 #1296: [moveit_configs_utils] Update to use standard warehouse launch

DLu, 2022-05-31:
[This function](https://github.com/ros-planning/moveit2/blob/934ac23a3c0834df611a762073c0a514fb96fe86/moveit_configs_utils/moveit_configs_utils/launches.py#L133-L184) uses `warehouse_ros_mongo` but [based on this conversation](https://github.com/ros-planning/moveit2/pull/1189#discussion_r853145289) it should probably be `warehouse_ros_sqlite`. 

(I wrote this code based on the ROS 1 launch files, but have never used the warehouse functionality, so it would be good for someone with passing knowledge of warehouses to test it)

henningkayser, 2023-01-26:
Ok, it's really overdue to get this done. MongoDB is not provided with Jammy anymore, so we should:
* Deprecate the old package https://github.com/ros-planning/warehouse_ros_mongo/issues/71
* Update launch code, links and build badges

mannygarcia720, 2023-06-05:
Hello! I am interested on working on this issue. Is there anything I should know beforehand in order to get started on solving this issue? @DLu @henningkayser

jorgenicho-brt, 2023-07-13:
I recently started to use the warehouse_ros database capability and got something that saves constraints to a sqlite database and also have a launch configuration that loads the path constraints from the db from the rviz motion planning frame.  
However it is unclear to me if the move_group node loads the database whenever provided the database parameters (warehouse_host and warehouse_plugin).  It appears that it doesn't at the moment but it feels to me that it should do that upon request.  I could add this feature to the move_group node if it makes sense, I think the move_group node in moveit(1) did this.

jrgnicho, 2023-07-13:
I recently started to use the warehouse_ros database capability and got something that saves constraints to a sqlite database and also have a launch configuration that loads the path constraints from the db through the rviz motion planning frame.  
However it is unclear to me if the move_group node loads the database whenever provided the database parameters (warehouse_host and warehouse_plugin).  It appears that it doesn't at the moment but it feels to me that it should do that upon request.  I could add this feature to the move_group node if it makes sense, I think the move_group node in moveit(1) did this.