moveit/moveit2 #1298: Program stuck creating moveit::planning_interface::MoveGroupInterface object.

Cryoschrome, 2022-06-01:
### Environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source build - humble branch (6288b0d2)

### Description
Hi,
I have been trying to work with move_group API, and created a [sample project](https://github.com/Cryoschrome/moveit_test_pkg).
When launching the file using [new_test.launch.py](https://github.com/Cryoschrome/moveit_test_pkg/blob/master/launch/new_test.launch.py). The program is stuck while creating the moveit::planning_interface::MoveGroupInterface object ([line 27](https://github.com/Cryoschrome/moveit_test_pkg/blob/362c0f1f7cf73d1ebca0b5cee88fa9a53c0134f9/src/move_group_test.cpp#L27)). The program does not provide any errors.

i get the following output.
```
cryoschrome@ROG:~/grab_ws$ ros2 launch moveit_test_pkg new_test.launch.py 
[INFO] [launch]: All log files can be found below /home/sourabh/.ros/log/2022-06-01-11-52-11-019536-ROG-8627
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [8628]
[INFO] [move_group_test-2]: process started with pid [8630]
[robot_state_publisher-1] Link rotate_joint had 0 children
[robot_state_publisher-1] [INFO] [1654069931.256974968] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1654069931.257213403] [robot_state_publisher]: got segment rotate_joint
[move_group_test-2] [INFO] [1654069931.293252025] [move_group_test]: MOVE_GROUP CREATED.
[move_group_test-2] [INFO] [1654069931.293500727] [move_group_test]: EXECUTOR Added to thread
[move_group_test-2] [INFO] [1654069931.301201189] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00286066 seconds
[move_group_test-2] [INFO] [1654069931.301249099] [moveit_robot_model.robot_model]: Loading robot model 'single_joint'...
[move_group_test-2] Link rotate_joint had 0 children
```

### Expected behaviour
moveit::planning_interface::MoveGroupInterface object is created and the rest of the code is executed.

### Actual behaviour
Stuck creating moveit::planning_interface::MoveGroupInterface object

henningkayser, 2022-06-02:
@Cryoschrome could you try if your issue still exists with #1305?

henningkayser, 2022-06-02:
Oh, I see. It seems like you're not launching the MoveGroup node. Running the MoveGroupInterface without another MoveGroup node doesn't work, as it's merely a ROS interface. Please review the [MoveGroupInterface tutorial](https://moveit.picknik.ai/humble/doc/examples/move_group_interface/move_group_interface_tutorial.html). You'll see that the launch files include separate entries for [MoveGroup](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/move_group_interface/launch/move_group.launch.py#L16) and [MoveGroupInterface](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/move_group_interface/launch/move_group_interface_tutorial.launch.py). Also, there are several other nodes that you might want to launch as well (e.g. controllers), depending on your needs. It's also highly recommended to use [moveit_configs_utils](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/move_group_interface/launch/move_group.launch.py#L11) for generating the launch parameters, it makes your life a lot easier.

closed 2022-06-02 by henningkayser