moveit/moveit2 #1303: Exception when using persistent roadmaps

brownbrowny, 2022-06-02:
### Description

I am currently using Moveit2 in combination with ROS Foxy to control a Universal Robot UR10e. The purpose is to try different planning algorithms for trajectory generation and validate them.
So far everything worked fine more or less (had some build problems). Then I found the feature of persistent roadmaps which came with OMPL 1.5.0 which I wanted to try of course. After adapting the ompl_planning.yaml accordingly I tried planning with the new planner but got a screen full of exceptions produced by Moveit.

This is my first report, so please excuse any mistakes or missing information and let me know. Also tips for improvements are welcome.

### My environment
* ROS Distro: Foxy
* OS Version: Ubuntu 20.04.4 LTS
* Source or Binary build?: Source
* If source, which branch?: https://github.com/ros-planning/moveit2/releases/tag/2.2.2

### Steps to reproduce
1. For using the planners with persistent roadmaps I followed this tutorial: https://moveit.picknik.ai/humble/doc/examples/ompl_interface/ompl_interface_tutorial.html#persistent-roadmaps
So my ompl_planning.yaml looks like:
`planner_configs:
  PersistentLazyPRMstar: # use this with a representative environment to create a roadmap
    type: geometric::LazyPRMstar
    multi_query_planning_enabled: true
    store_planner_data: true
    load_planner_data: false
    planner_data_path: /tmp/roadmap.graph
  PersistentLazyPRM: # use this to load a previously created roadmap
    type: geometric::LazyPRM
    multi_query_planning_enabled: true
    store_planner_data: false
    load_planner_data: true
    planner_data_path: /tmp/roadmap.graph
  SemiPersistentLazyPRMstar: # reuses roadmap during lifetime of node but doesn't save/load roadmap to/from disk
    type: geometric::LazyPRMstar
    multi_query_planning_enabled: true
    store_planner_data: false
    load_planner_data: false
  SemiPersistentLazyPRM: # reuses roadmap during lifetime of node but doesn't save/load roadmap to/from disk
    type: geometric::LazyPRM
    multi_query_planning_enabled: true
    store_planner_data: false
    load_planner_data: false
ur_manipulator:
  planner_configs:
    - PersistentLazyPRMstar
    - PersistentLazyPRM
    - SemiPersistentLazyPRMstar
    - SemiPersistentLazyPRM`
    - 
2. Rebuilding the corresponding package ur_moveit_config

3. Start my Robot Control driver in simulation mode which I have from: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e use_fake_hardware:=true launch_rviz:=false'`
4. Start Moveit with Rviz:
`ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e use_fake_hardware:=true launch_rviz:=true'`

### Expected behaviour
- I should be able to select the four planners I configured in the ompl_planning.yaml
- Then when planning to a goal pose by using the interactive marker, the planner should create a plan and visualize the computed
trajectory

### Actual behaviour
- I can see the planners in the dropdown in the Context tab, also selecting works. Then I can see the correct planner parameters
- As soon as I press Plan with default options, nothing happens but the error log in the Moveit console window
- Other planners without persistent roadmaps work fine
- 
### Backtrace or Console output

<script src="https://gist.github.com/brownbrowny/b025a0cab6a7b3ddcc0b269c51ddb0af.js"></script>

If you wonder why there are no other planners in the ompl_planning.yaml, I removed them all just to be sure there is no error there. They work without any problems at all.

I also just kept:
`
PersistentLazyPRMstar: # use this with a representative environment to create a roadmap
    type: geometric::LazyPRMstar
    multi_query_planning_enabled: true
`
But the error still occurs, so it must be the option _multi_query_planning_enabled_

Any ideas?

FarisHamdi, 2022-06-25:
I am also facing a similar issue, but while attempting to use AnytimePathShortening

FarisHamdi, 2022-06-26:
I figured it out! 

Config parsing in the planning pipeline is unable to parse `true` or `false`, due to the lexical cast in the line below. This issue affects any feature that were previously enabled via booleans including multi_query_planning
https://github.com/ros-planning/moveit2/blob/949facee1889bf2ba2dd76711051ab01bd307f51/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp#L359

This can be gotten around by using `1` for true or `0` for false

AndyZe, 2022-07-06:
The documentation was fixed here:  https://github.com/ros-planning/moveit2_tutorials/pull/460

I'm going to close the issue now. Thanks for tracking this down!

closed 2022-07-06 by AndyZe

brownbrowny, 2022-07-07:
Sorry for replying so late. I tried your solution @FarisHamdi and it works perfectly. Thank you!