moveit/moveit2 #1314: Test Windows Build in CI

ChrisThrasher, 2022-06-02:
MoveIt already provides [binaries for Windows](https://moveit.ros.org/install-moveit2/binary-windows/) so it's in our best interest to continuously build and test on Windows to ensure we don't introduce Windows regressions. Hopefully there's a precedent for doing Windows ROS 2 CI actions, but if not then we might have to establish those conventions. 

@lilustga Do you know how we might approach this?

lilustga, 2022-06-03:
Hi @ChrisThrasher,

Moveit CI for Windows would be great! There's definitely precedent for Windows ROS 2 CI.

The Github action is likely the best place to start: https://github.com/ros-tooling/action-ros-ci

v4hn, 2022-06-07:
Should this be based [on the RoboStack CI](https://github.com/ros-planning/moveit/pull/2604)?

closed 2024-11-08 by ChrisThrasher