moveit/moveit2 #1317: Build from Source Moveit2  fails- ROS Humble

Nils-ChristianIseke, 2022-06-03:
### Description

Following the tutorial: https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html, the build fails with the following error:
```
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)

```
Also the following warning is shown:
` WARNING:colcon.colcon_ros.prefix_path.ament:The path '/opt/ros/rolling' in the environment variable AMENT_PREFIX_PATH doesn't exist`

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build: ROS2 Humble -> Binary, Moveit2 Humble -> Source
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Follow the tutorial https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html
### Expected behaviour
Build succeds.

### Actual behaviour
Build fails.

### Backtrace or Console output
```
colcon build --mixin release
[0.952s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/opt/ros/rolling' in the environment variable AMENT_PREFIX_PATH doesn't exist
Starting >>> srdfdom 
Starting >>> moveit_common
Starting >>> moveit_resources_panda_description
Starting >>> launch_param_builder
Starting >>> moveit_resources_fanuc_description                               
Starting >>> moveit_task_constructor_msgs
Starting >>> rviz_marker_tools
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_panda_description [1.01s]           
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_common [1.09s]
Finished <<< moveit_resources_fanuc_description [1.02s]        
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_resources_pr2_description [0.98s]
Starting >>> moveit_resources_fanuc_moveit_config
Starting >>> rosparam_shortcuts
Finished <<< rviz_marker_tools [1.20s]                         
Finished <<< srdfdom [1.42s]
--- stderr: moveit_task_constructor_msgs                            
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)
gmake[2]: *** [moveit_task_constructor_msgs__py/CMakeFiles/moveit_task_constructor_msgs__py.dir/build.make:328: rosidl_generator_py/moveit_task_constructor_msgs/_moveit_task_constructor_msgs_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:590: moveit_task_constructor_msgs__py/CMakeFiles/moveit_task_constructor_msgs__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_msgs [1.42s, exited with code 2]
Aborted  <<< moveit_resources_prbt_support [0.41s]
Aborted  <<< moveit_resources_panda_moveit_config [0.46s]
Aborted  <<< moveit_resources_fanuc_moveit_config [0.36s]
Aborted  <<< launch_param_builder [1.60s]         
Aborted  <<< rosparam_shortcuts [1.90s]                             

Summary: 6 packages finished [4.28s]
  1 package failed: moveit_task_constructor_msgs
  5 packages aborted: launch_param_builder moveit_resources_fanuc_moveit_config moveit_resources_panda_moveit_config moveit_resources_prbt_support rosparam_shortcuts
  2 packages had stderr output: launch_param_builder moveit_task_constructor_msgs
  38 packages not processed

```

Nils-ChristianIseke, 2022-06-03:
Thank you for your awnser. Don't know how thtat happened :p. 
A clean build resolves the warning. But does not resolve the following errors:
```
--- stderr: moveit_configs_utils        
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [1.87s]
--- stderr: moveit_task_constructor_msgs
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)
```

JafarAbdi, 2022-06-03:
What's your `PYTHONPATH` env-variable value?

GradyM2M, 2022-06-08:
@Nils-ChristianIseke Hi, I have the same issue, any progress on this?

Nils-ChristianIseke, 2022-06-08:
For now, i decided to switch back to Ubuntu 20.04 and ROS Foxy at work, because i also had some trouble with Ubuntu 22.04. But I will try to reproduce this error on my private pc, as soon as i have some spare time.

vatanaksoytezer, 2022-06-24:
Anyone still having this issue? The CI, my local installation and dockers have been pretty happy for some time.

GradyM2M, 2022-06-27:
When I try to reinstall ros humble, the issue is gone. LOL

vatanaksoytezer, 2022-06-27:
Closing the issue, then. Please feel free to re-open or open a new one, if you have other issues.

closed 2022-06-27 by vatanaksoytezer