moveit/moveit2 #1334: RDF Loader Flaky

henrygerardmoore, 2022-06-10:
### Description

The RDF Loader appears to fail in CI for no reason on [one commit](https://github.com/ros-planning/moveit2/pull/1331/commits/58f3828224f051b40ce0b66d217ae4335703a5d2) and then succeeds on [the other](https://github.com/ros-planning/moveit2/pull/1331/commits/190aa61aa48bd504b569767e27410f907339eeca) where only a comment was removed.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Built from source on the remove_boost branch of my fork (gh pr checkout 1331 to see)

### Steps to reproduce
Only encountered this once, so it is difficult to say how to reproduce it. Possibly just repeated runs of CI will do it.

### Expected behaviour
Test should pass consistently

### Actual behaviour
Test appears to fail inconsistently

### Backtrace or Console output

[CI for failing test](https://github.com/ros-planning/moveit2/runs/6821795888?check_suite_focus=true)
[CI for passing test](https://github.com/ros-planning/moveit2/runs/6834209568?check_suite_focus=true)

mikeferguson, 2022-07-01:
I notice in the logs "terminate called without an active exception" - which usually means that a std::thread went out of scope while not joined/stopped - I notice we have a std::thread in this test - but I don't actually see how it could go out of scope...

I fixed an instance of this in https://github.com/ros-planning/moveit2/pull/1384

henningkayser, 2023-01-26:
I haven't seen this one lately. Are we still running into this? Maybe it really was an unrelated threading issue somewhere...

vatanaksoytezer, 2023-01-26:
I think I saw this like a few months ago, but haven't seen lately.

henningkayser, 2023-03-10:
closing for now

closed 2023-03-10 by henningkayser