moveit/moveit2 #1335: Servo test fails locally

henrygerardmoore, 2022-06-10:
### Description

The test_servo_integration test fails for me on main as well as on my own branch (remove_boost) when run locally. However, it does not cause CI to fail on PRs.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Built from source on main branch

### Steps to reproduce
Run colcon test in moveit2 work space. Note that the servo test hangs for a long time before failing. Run test_servo_integration directly with `./build/moveit_servo/test_servo_integration` and see the output.
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Test should pass locally or cause CI to fail.

### Actual behaviour
Test times out when run with colcon test, runs indefinitely when run directly

### Backtrace or Console output

[Gist with direct run, colcon test, and colcon test result outputs](https://gist.github.com/henrygerardmoore/9423d788103b6d57afb917f909fe70e9)

AndyZe, 2022-06-11:
Well, I can't reproduce this locally. All Servo tests pass for me. But that doesn't mean anything. I'll push a branch for you to test that might help with a race condition.

AndyZe, 2022-06-13:
That's too bad. I was hoping it would be the easy fix  :(

I've seen @griswaldbrooks  do this, too. I don't know too much about it:

namespace {

```
// This is a bit of a hack to make thread sanitizer ignore a race condition
// in the constructor of the rclcpp::Node
#if defined(__has_feature)
#if __has_feature(thread_sanitizer)
__attribute__((no_sanitize("thread")))
#endif
#endif
rclcpp::Node::SharedPtr
make_node(std::string const& name, rclcpp::NodeOptions const& options) {
  return std::make_shared<rclcpp::Node>(name, options);
}
```

^This goes near the top of the test cpp source file. Then later, make_node is used like this:

`const auto node = make_node("planner_test", rclcpp::NodeOptions{});`

closed 2022-06-13 by henrygerardmoore

griswaldbrooks, 2022-06-14:
> That's too bad. I was hoping it would be the easy fix :(
>
> I've seen @griswaldbrooks do this, too. I don't know too much about it:
>
> namespace {
>
> ```
> // This is a bit of a hack to make thread [...]

@AndyZe this was originally from work done by @tylerjw and @jliukkonen

tylerjw, 2022-06-15:
Looking at the output, I would guess this has something to do with the ros2_control version or moveit_resources version you are testing with vs what CI is testing with. 

Could you post the result of `vcs export` in the src directory of your workspace?

Secondly, what version of Ubuntu and what version of ROS are you testing on?

I assume you've updated rosdep and apt and done an apt-upgrade recently, but just to be sure, try that.

@AndyZe that code is to try to disable thread sanitizer checking of the node constructor if you are building with thread sanitizer. As the constructor for Node has race conditions it finds that code helped us use tsan in our unit tests. Sadly though there are new race conditions in ROS that broke our tests so for now we have disabled tsan testing in our CI for now in the project where we wrote that.

henrygerardmoore, 2022-06-15:
`repositories:
  launch_param_builder:
    type: git
    url: https://github.com/PickNikRobotics/launch_param_builder
    version: main
  moveit2:
    type: git
    url: https://github.com/henrygerardmoore/moveit2
    version: main
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources
    version: ros2
  moveit_task_constructor:
    type: git
    url: https://github.com/ros-planning/moveit_task_constructor.git
    version: ros2
  moveit_visual_tools:
    type: git
    url: https://github.com/ros-planning/moveit_visual_tools
    version: ros2
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control.git
    version: master
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers.git
    version: master
  rosparam_shortcuts:
    type: git
    url: https://github.com/PickNikRobotics/rosparam_shortcuts
    version: ros2
  srdfdom:
    type: git
    url: https://github.com/ros-planning/srdfdom.git
    version: ros2`
I also updated rosdep and apt update and upgrade right before this.

PR #1336 (2022-06-14): Update Servo ros2_control launching
### Description

May fix #1335.

This updates the ros2_controller launching to be up-to-date with the latest.

I used the same technique in this moveit2_tutorials PR:  https://github.com/ros-planning/moveit2_tutorials/pull/389