moveit/moveit2 #1337: Pilz execution fails on 'invalid transition from state EXECUTING with event EXECUTE'

JensVanhooydonck, 2022-06-11:
### Description

It seems that the state change is not possible.  I thinks this only happens when using a path sequence.
Also an empty callback is not working and crashes move_group.. 

Changing `goal_handle->execute();` to
```  
if(!goal_handle->is_executing()) {
    goal_handle->execute();
}
```
Fixes that part. 
Commenting out

```
  // move_feedback_->state = stateToStr(state);
  // goal_handle_->publish_feedback(move_feedback_);
``` 
in `move_group_sequence_action.cpp` makes it execute instead of crashing move_group.

### Your environment
* ROS Distro: Galactic
* OS Version: Ubuntu 18.04 (docker)
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch? galactic

### Steps to reproduce
Plan & Executing a pilz path.

### Expected behaviour
Execute the path.

### Actual behaviour
move_group crashes.

### Backtrace or Console output
```
[ros2_control_node-5] [INFO] [1654934791.258254779] [tmr_arm_controller]: Received new action goal
[ros2_control_node-5] [INFO] [1654934791.258351335] [tmr_arm_controller]: Accepted new action goal
[move_group-3] [INFO] [1654934791.263505634] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: tmr_arm_controller started execution
[move_group-3] [INFO] [1654934791.263650277] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-5] [INFO] [1654934791.726585969] [tmr_arm_controller]: Goal reached, success!
[move_group-3] [INFO] [1654934791.731172161] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'tmr_arm_controller' successfully finished
[move_group-3] [INFO] [1654934791.800205294] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-3] [INFO] [1654934791.800544561] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: SUCCEEDED
[move_group-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[move_group-3]   what():  goal_handle attempted invalid transition from state EXECUTING with event EXECUTE, at /tmp/binarydeb/ros-galactic-rcl-action-3.1.3/src/rcl_action/goal_handle.c:95
[move_group-3] Stack trace (most recent call last) in thread 3654:
[move_group-3] #20   Object "", at 0xffffffffffffffff, in 
[server-4] [INFO] [1654934791.843394278] [ign_moveit2_py]: goal send
[server-4] [INFO] [1654934791.845553470] [Server]: GOAL: ClientGoalHandle <id=[145 140 148  25  93  97  70 178 144  37  46 223 231  92  67 233], accepted=True, status=2>
[move_group-3] #19   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in __clone [0x7fa3ced28132]
[move_group-3] #18   Source "/build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7fa3cd6c1608]
[move_group-3] #17   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fa3cf020de3, in 
[move_group-3] #16   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa3cf2374ba, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[move_group-3] #15   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa3cf2311b2, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[move_group-3] #14   Object "/opt/ros/galactic/lib/librclcpp_action.so", at 0x7fa3cd37beef, in rclcpp_action::ServerBase::execute_goal_request_received(std::shared_ptr<void>&)
[move_group-3] #13   Object "/workspaces/ws_robot/install/lib/libsequence_capability.so", at 0x7fa3a45e0605, in rclcpp_action::Server<moveit_msgs::action::MoveGroupSequence>::call_goal_accepted_callback(std::shared_ptr<rcl_action_goal_handle_t>, std::array<unsigned char, 16ul>, std::shared_ptr<void>)
[move_group-3] #12   Object "/workspaces/ws_robot/install/lib/libsequence_capability.so", at 0x7fa3a45e1890, in std::function<void (std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >)>::operator()(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >) const
[move_group-3] #11   Object "/workspaces/ws_robot/install/lib/libsequence_capability.so", at 0x7fa3a45aa209, in std::_Function_handler<void (std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >), pilz_industrial_motion_planner::MoveGroupSequenceAction::initialize()::{lambda(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >)#3}>::_M_invoke(std::_Any_data const&, std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >&&)
[move_group-3] #10   Object "/workspaces/ws_robot/install/lib/libsequence_capability.so", at 0x7fa3a45a69d8, in pilz_industrial_motion_planner::MoveGroupSequenceAction::initialize()::{lambda(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >)#3}::operator()(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >) const
[move_group-3] #9    Object "/workspaces/ws_robot/install/lib/libsequence_capability.so", at 0x7fa3a45a701e, in pilz_industrial_motion_planner::MoveGroupSequenceAction::executeSequenceCallback(std::shared_ptr<rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence> >)
[move_group-3] #8    Object "/workspaces/ws_robot/install/lib/libsequence_capability.so", at 0x7fa3a45b875d, in rclcpp_action::ServerGoalHandle<moveit_msgs::action::MoveGroupSequence>::execute()
[move_group-3] #7    Object "/opt/ros/galactic/lib/librclcpp_action.so", at 0x7fa3cd37f797, in rclcpp_action::ServerGoalHandleBase::_execute()
[move_group-3] #6    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa3cf22ab5c, in rclcpp::exceptions::throw_from_rcl_error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcutils_error_state_t const*, void (*)())
[move_group-3] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fa3ceff437e, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
[move_group-3] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fa3ceff43f6, in std::terminate()
[move_group-3] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fa3ceff438b, in 
[move_group-3] #2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fa3cefe8910, in 
[move_group-3] #1    Source "/build/glibc-SzIz7B/glibc-2.31/stdlib/abort.c", line 79, in abort [0x7fa3cec2b858]
[move_group-3] #0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7fa3cec4c00b]
[move_group-3] Aborted (Signal sent by tkill() 3548 1000)
```
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

JensVanhooydonck, 2022-06-13:
@JafarAbdi As requested

henningkayser, 2023-03-10:
@JafarAbdi ping