moveit/moveit2 #1358: Moveit2 Setup Assistant

MYusernameK, 2022-06-14:
I want to use the Movit2 Setup Assistant. I prefer it in foxy, but can't install it with sudo apt install, so I installed gazebo and the with apt the moveit setup assistant. With "ros2 launch moveit_setup_assistant setup_assistant.launch"  I get:

`[INFO] [launch]: All log files can be found below /home/kuka/.ros/log/2022-06-14-14-06-57-215759-kuka-MacBookPro-97826
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: The launch file may have a syntax error, or its format is unknown
`

I can't figure out how to use the Assistant. Can you please tell me?

vatanaksoytezer, 2022-06-14:
Thank you for opening an issue. MSA will not be available for distributions before Humble including Foxy and Galactic. Having said that, MSA is not fully ready right now, and we are working on getting it to Humble and Rolling as soon as possible. In the meantime either try to generate the required files by hand or use the ROS 1 version and modify files. Sorry for any inconvenience. I'm going to close this issue since it's answered, please feel free to re-open or continue the conversation.

closed 2022-06-14 by vatanaksoytezer

connoranderson, 2022-08-23:
Also ran into this. Would be great if galactic documentation removed reference to the tool if it's not supported.

https://moveit.picknik.ai/galactic/doc/examples/setup_assistant/setup_assistant_tutorial.html