moveit/moveit2 #1379: Missing ros2 py lunch file in setup assistant

yoavRobee, 2022-06-19:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* Ubuntu 22
* Source build
* If source, which branch? Humbale -stable

The setups assistant package includes still the old ROS  lunch file

mikeferguson, 2022-06-20:
You might be interested in trying this feature branch: https://github.com/ros-planning/moveit2/pull/1254 - the MSA is still under heavy development

vatanaksoytezer, 2022-06-20:
New MSA has not been merged and is in the `feature/msa` branch as @mikeferguson suggested. I'm going to close this issue since the branch hasn't been merged but please feel free to follow up on #1254 or open a new issue if you test the branch.

closed 2022-06-20 by vatanaksoytezer