moveit/moveit2 #1390: MSA: joints used by multiple groups are duplicated in ros2_control.xacro

mikeferguson, 2022-06-26:
### Description

Some robots have multiple groups with similar joints (for instance: arm, arm_with_torso). When this is configured in the MSA, the joints that are present in both groups show up twice in the ros2_control.xacro

### Your environment
* ROS Distro: humble
* OS Version: Ubuntu 22.04
* Source build of feature/ms

mikeferguson, 2022-06-26:
FYI @DLu

DLu, 2022-06-27:
There is [a bug where those joints are added to the `joint_names_` vector twice](https://github.com/ros-planning/moveit2/blob/6ab20361242cc81dc06913f3ac52486d31c11f5e/moveit_setup_assistant/moveit_setup_controllers/src/control_xacro_config.cpp#L128) but because the generated interfaces are [`unordered_map` based](https://github.com/ros-planning/moveit2/blob/6ab20361242cc81dc06913f3ac52486d31c11f5e/moveit_setup_assistant/moveit_setup_controllers/include/moveit_setup_controllers/control_xacro_config.hpp#L191), I'm currently not seeing the duplicated values in `ros2_control.xacro`. 

Are you using the MoveIt config [checked  in here](https://github.com/mikeferguson/ubr_reloaded/tree/ros2/ubr1_moveit), @mikeferguson ?

mikeferguson, 2022-06-27:
That is the latest export (23 hours ago from now) and if you see https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_moveit/config/ubr1.ros2_control.xacro it has duplicated data

mikeferguson, 2022-06-29:
The issue appears to be that we do the export by looping over the joint_names_ (which has duplicates): https://github.com/ros-planning/moveit2/blob/6ab20361242cc81dc06913f3ac52486d31c11f5e/moveit_setup_assistant/moveit_setup_controllers/src/control_xacro_config.cpp#L233 - PR incoming

PR #1406 (2022-07-01): msa: fix duplicated ros_control joints
### Description

When more than one controller can control a joint, that joint would appear multiple times. Fixes #1390

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit/blob/master/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [refere [... cut]
closed 2022-07-01 by henningkayser