moveit/moveit2 #1393: MoveitConfigBuilder: capabilities is unused

mikeferguson, 2022-06-26:
### Description

move_group_capabilities is defined at https://github.com/ros-planning/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py#L106 - but is never actually filled in or loaded from a yaml file

There isn't a clear way to define the loading of custom capabilities (for instance the Task Constructor ones).

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04

vatanaksoytezer, 2022-06-28:
> Should we load this from YAML?

Loading from YAML looks like the correct way to me. MSA / MoveIt 2 launches are generally structured such that you don't need to edit launch files. @JafarAbdi thoughts?

henningkayser, 2023-03-10:
@JafarAbdi ping

chama1176, 2023-07-18:
Hi, I have same problem as this comment https://github.com/ros-planning/moveit2/issues/1248#issuecomment-1350773802
, and it is related to this issue.

`MoveItConfigsBuilder` and `generate_move_group_launch` load `planning_pipelines` `capabilities` param into the `/move_group pilz_industrial_motion_planner.capabilities` (or something like this).

However, move_group loads pipeline_capabilities from `"planning_pipelines/" + pipeline_name + "/capabilities"`.
This means `/move_group planning_pipelines/pilz_industrial_motion_planner/capabilities` and this param path is invalid in ros2 and cannot load any capabilities.
https://github.com/ros-planning/moveit2/blob/b0f0f680c3f86b8074d208a1e78c92cfa75cf5ca/moveit_ros/move_group/src/move_group.cpp#L155-L161

So, https://github.com/ros-planning/moveit2/issues/1248#issuecomment-1350773802 uses `capabilities(not related to any pipelines)` to load capabilities. 

Is there any solution?
Or I fixed it by changing the path as follows. Is it ok?
```
      if (context_->moveit_cpp_->getNode()->get_parameter(pipeline_name + ".capabilities",
                                                          pipeline_capabilities))
```

chama1176, 2023-07-31:
Thanks for the #2270 , my problem is solved.
https://github.com/ros-planning/moveit2/issues/1393#issuecomment-1639516823

mikeferguson, 2024-10-29:
This is actually now supported by modifying the launch file, which is what I've done here: https://github.com/mikeferguson/ubr_reloaded/commit/84c67efae0a58883bc21765859f968ae2c29fc2f