moveit/moveit2 #1394: moveit_simple_controller_manager cannot find controller.

Cryoschrome, 2022-06-27:
### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Built type: Source Build
* Branch: Humble

### Description

Hi, I am trying to understand and work with MoveIt for custom robots. I have created files for a simple model and currently trying to implement the low-level controller. I have created a hardware-interface for `ros2_controller` - `JointTrajectoryController.` While running the demo, when I try to execute the plan I am getting errors. 
The `.yaml` files are given below:
#### moveit_controller.yaml
```
trajectory_execution:
  allowed_execution_duration_scaling: 1.2
  allowed_goal_duration_margin: 0.5
  allowed_start_tolerance: 0.01

moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - rotate_joint_controller
  
  rotate_joint_controller:
    actions_ns: follow_joint_trajectory
    type: FollowJointTrajectory
    default: true
    joints:
      - base_to_rotate_joint
```
#### ros_controllers.yaml
```
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    rotate_joint_controller:
      type: "joint_trajectory_controller/JointTrajectoryController"

rotate_joint_controller:
  ros__parameters:
    joints:
      - base_to_rotate_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
```
I launch move_group node using the following launch file.
#### Launch file
```
import os
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import ExecuteProcess
from ament_index_python.packages import get_package_share_directory
from moveit_configs_utils import MoveItConfigsBuilder

def generate_launch_description():
  moveit_config = (
    MoveItConfigsBuilder("one_joint")
    .robot_description(file_path="config/one_joint.urdf.xacro")
    .trajectory_execution(file_path="config/moveit_controllers.yaml")
    .to_moveit_configs()
  )

  run_move_group_node = Node(
    package="moveit_ros_move_group",
    executable="move_group",
    output="screen",
    parameters=[moveit_config.to_dict()],
  )

  # rviz_config_file = (
  #       get_package_share_directory("moveit2_tutorials") + "/launch/move_group.rviz"
  #   )
  rviz_node = Node(
    package="rviz2",
    executable="rviz2",
    name="rviz2",
    output="log",
    #arguments=["-d", rviz_config_file],
    parameters=[
        moveit_config.robot_description,
        moveit_config.robot_description_semantic,
        moveit_config.robot_description_kinematics,
        moveit_config.planning_pipelines,
        moveit_config.joint_limits,
    ],
  )

  static_tf = Node(
    package="tf2_ros",
    executable="static_transform_publisher",
    name="static_transform_publisher",
    output="log",
    arguments=["0.0", "0.0", "0.0", "0.0", "0.0", "0.0", "world", "base_link"],
  )

  # Publish TF
  robot_state_publisher = Node(
    package="robot_state_publisher",
    executable="robot_state_publisher",
    name="robot_state_publisher",
    output="both",
    parameters=[moveit_config.robot_description],
  )

  ros2_controllers_path = os.path.join(
    get_package_share_directory("one_joint_moveit_config"),
    "config",
    "ros_controllers.yaml",
  )
  ros2_control_node = Node(
    package="controller_manager",
    executable="ros2_control_node",
    parameters=[moveit_config.robot_description, ros2_controllers_path],
    output="screen",
  )

  load_controllers = []
  for controller in [
    "rotate_joint_controller",
  ]:
    load_controllers += [
      ExecuteProcess(
        cmd=["ros2 run controller_manager spawner {}".format(controller)],
        shell=True,
        output="screen",
      )
    ]
  print(load_controllers);
  return LaunchDescription(
    [
      rviz_node,
      static_tf,
      robot_state_publisher,
      run_move_group_node,
      ros2_control_node
    ]
    + load_controllers
  )
```
When executing the plan I get the following error
```
[move_group-4] [INFO] [1656331116.837111428] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-4] [INFO] [1656331116.837433043] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-4] [INFO] [1656331116.837514826] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-4] [INFO] [1656331116.837546324] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-4] [INFO] [1656331116.837585365] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-4] [INFO] [1656331116.837602615] [moveit.plugins.moveit_simple_controller_manager]: Returned 0 controllers in list
[move_group-4] [ERROR] [1656331116.837633624] [moveit_ros.trajectory_execution_manager]: Unable to identify any set of controllers that can actuate the specified joints: [ base_to_rotate_joint ]
[move_group-4] [ERROR] [1656331116.837655554] [moveit_ros.trajectory_execution_manager]: Known controllers and their joints:
[move_group-4] 
```

The `rotate_joint_controller` is up and running. I checked the parameter using `ros2 param` and received the following ouput.
```
┌─[cryos][ROG][~]
└─➞ ros2 param get /moveit_simple_controller_manager moveit_simple_controller_manager.controller_names 
String values are: ['rotate_joint_controller']
┌─[cryos][ROG][~]
└─➞ ros2 param get /moveit_simple_controller_manager moveit_simple_controller_manager.rotate_joint_controller.joints 
String values are: ['base_to_rotate_joint']
```

### Expected behaviour
The trajectory is executed

### Actual behaviour
MoveIt simple controller manager cannot find the controllers for the joint

Any insight on this issue in appreciated.
Thank you.

Cryoschrome, 2022-06-29:
Thanks for the reply!
Will you the `feature/msa` branch moving forward.
For now, I got it working by setting `moveit_controller_manager: moveit_ros_control_interface/MoveItControllerManager` in `moveit_controllers.yaml`

closed 2022-06-29 by Cryoschrome

vatanaksoytezer, 2022-06-29:
@Cryoschrome just a small update feature/msa has been merged to main, so feel free to use main branch going forward