moveit/moveit2 #1399: Check clock synchronization problem with getCurrentState()

T2FF, 2022-06-28:
Hello,
I am using moveit2 in combination with the [Universal_Robots_ROS2_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/foxy) and Foxy distribution. Unfortunately I found out that there is no Python interface for Move-Group (only C++), which is a problem for me as my previous project is completely based on Python. However, since my task only involves Pick&Place, I wanted to write an additional node in C++ with a very simple service that I can control via Python (rclpy). The service should receive a new position with orientation and move there from the current position of the robot via a cartesian path. I followed the [Moveit2 tutorial](https://github.com/ros-planning/moveit2_tutorials/blob/foxy/doc/move_group_interface/src/move_group_interface_tutorial.cpp) as far as possible and defined my additional node as an ROS package with its own service:

```
#include <memory>
#include <iostream>
#include <typeinfo>
#include <unistd.h>
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include "moveit2_interface/srv/move_cartesian.hpp"

using moveit::planning_interface::MoveGroupInterface;
MoveGroupInterface *move_group_interface;

void move_cartesian(const std::shared_ptr<moveit2_interface::srv::MoveCartesian::Request> request, std::shared_ptr<moveit2_interface::srv::MoveCartesian::Response> response)  // CHANGE
{
  response->success = true;
  response->notsuccess = false;
  RCLCPP_INFO(rclcpp::get_logger("moveit2_interface"), "Incoming request\no_w: %1.2lf" "\no_x: %1.2lf" "\no_y: %1.2lf" "\no_z: %1.2lf" "\np_x: %1.2lf" "\np_y: %1.2lf" "\np_z: %1.2lf",request->orientation_w, request->orientation_x, request->orientation_y, request->orientation_z, request->position_x, request->position_y, request->position_z);
  RCLCPP_INFO(rclcpp::get_logger("moveit2_interface"), "sending back response: %ld %ld", response->success, response->notsuccess);

  RCLCPP_INFO(rclcpp::get_logger("moveit2_interface"), "Read Robot");
  moveit::core::RobotState start_state(*move_group_interface->getCurrentState());
  move_group_interface->setStartState(start_state);
  geometry_msgs::msg::PoseStamped start_pose_Stamped = move_group_interface->getCurrentPose(); 
}

int main(int argc, char * argv[])
{

  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const move_group_node = std::make_shared<rclcpp::Node>("moveit2_interface_node", rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true));

  // We spin up a SingleThreadedExecutor for the current state monitor to get information about the robot's state.
  rclcpp::executors::SingleThreadedExecutor executor;
  executor.add_node(move_group_node);                       //Executer: https://docs.ros.org/en/foxy/Concepts/About-Executors.html
  std::thread thread([&executor]() { executor.spin(); });

  // Bind move_cartesian service to the node
  rclcpp::Service<moveit2_interface::srv::MoveCartesian>::SharedPtr service = move_group_node->create_service<moveit2_interface::srv::MoveCartesian>("move_cartesian", &move_cartesian);

  // Create a ROS logger
  RCLCPP_INFO(rclcpp::get_logger("moveit2_interface"), "Ready to receive commands");

  // Create the MoveIt MoveGroup Interface
  using moveit::planning_interface::MoveGroupInterface;

  // class can be easily set up using just the name of the planning group you would like to control and plan for.
  static const std::string PLANNING_GROUP = "ur_manipulator";
  move_group_interface = new MoveGroupInterface(move_group_node, PLANNING_GROUP);
   
  thread.join();
}
```
The node starts and runs without problems. However, when I address the service via Python and send it the new coordinates, they arrive but the node then hangs, apparently when trying to call the getCurrentState() method.
```
[moveit2interface-1] [INFO] [1656402904.697114816] [moveit2_interface]: Incoming request
[moveit2interface-1] o_w: 2.80
[moveit2interface-1] o_x: 1.10
[moveit2interface-1] o_y: 1.20
[moveit2interface-1] o_z: 1.30
[moveit2interface-1] p_x: 1.40
[moveit2interface-1] p_y: 1.50
[moveit2interface-1] p_z: 1.60
[moveit2interface-1] [INFO] [1656402904.697184885] [moveit2_interface]: sending back response: 1 0
[moveit2interface-1] [INFO] [1656402904.697200710] [moveit2_interface]: Lese Roboterachsen aus
[moveit2interface-1] [INFO] [1656402904.697941269] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[moveit2interface-1] [INFO] [1656402905.698064579] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[moveit2interface-1] Check clock synchronization if your are running ROS across multiple machines!
[moveit2interface-1] [ERROR] [1656402905.698140433] [move_group_interface]: Failed to fetch current robot state
[ERROR] [moveit2interface-1]: process has died [pid 8279, exit code -11, cmd '/home/thomas/workspace/ros_ur_driver/install/moveit2_interface/lib/moveit2_interface/moveit2interface --ros-args -r __node:=moveit2_interface --params-file /tmp/launch_params_np7j0a5c --params-file /tmp/launch_params_qwq9qd68 --params-file /tmp/launch_params_sqfepaaj'].
```
I have also found an entry that deals with a similar problem but without a solution: [Deal with ROS2 service & executors to get current robot states](https://github.com/ros-planning/moveit2/issues/496)
	
I don't quite understand where the problem lies here, since I use the executer. The problem also only occurs when I call getCurrentState() in my service function, in main it works without problems
Does anyone know how I can fix this problem?

Thanks in advance!

### Environment
* ROS Distro: Foxy
* OS Version: e.g. Ubuntu 20.04
* Binary Install

padhupradheep, 2022-09-22:
Was there any solution to this problem ?

henningkayser, 2023-01-26:
@T2FF did you manage to find the cause of your issue? There have been several fixes for executor behavior in the meantime, at least for the newer distros. From the log my first guess would have been that there is some kind of timing issue, maybe the service comes in before the first joint state, or the joint state topic is not publishing at all... It's also recommended to initially call [startStateMonitor()](https://moveit.picknik.ai/humble/api/html/classmoveit_1_1planning__interface_1_1MoveGroupInterface.html#ae9996710a32f8636c1bb82750929dca8) and to only start services after your node and objects have been initialized.

T2FF, 2023-01-26:
hi @henningkayser,
no unfortunately not I could not identify the problem more exactly at that time. I worked around the problem with global variables and did a cyclic query in the main function, after the initialization of the node, which reacted on the change of the global variables.

but thanks for the feedback!

henningkayser, 2023-01-26:
Thank you for the reply. If that worked, it pretty much confirms what I suggested: The state wasn't available for the service call yet. The call to starting the state monitor will also wait for a valid state before returning. I consider this solved for now, if you continue to have issues feel free to reopen.

closed 2023-01-26 by henningkayser