moveit/moveit2 #1407: Problem with MoveIt 2 Source Build - Linux

Llavrih, 2022-06-29:
## Description

I'm following steps on (https://moveit.ros.org/install-moveit2/source/) but cannot build Humble - stable. Whenever I try my terminal crashes.

## Your environment

    ROS Distro: humble
    OS Version: Ubuntu 22.04

vatanaksoytezer, 2022-06-29:
@Llavrih thank you for the issue, do you mind sharing what errors you are getting?

Llavrih, 2022-06-29:
The problem is that i don't even get any errors because my terminal just closes. This is the last pkg that starts building and than it crashes:
Starting >>> pilz_industrial_motion_planner
Sorry for the lack of info, I could send you log files.

Llavrih, 2022-06-29:
Thanks for the help anyway!

mikeferguson, 2022-06-29:
Almost certainly RAM related - try: "colcon build --executor sequential" - I've had that work just fine on 16GB of RAM

Llavrih, 2022-06-29:
Thanks @mikeferguson that worked like a charm.