moveit/moveit2 #1410: Release Rolling 2.6

henningkayser, 2022-06-30:
A new Rolling release is due by mid July, possibly also for Humble.
Rolling should be bumped to 2.6, Humble to 2.5.2.

vatanaksoytezer, 2022-06-30:
@henningkayser if we want we can cut it earlier, since MSA is merged and this would be the first release containing MSA.

vatanaksoytezer, 2022-06-30:
Also shouldn't we do 2.5.2 on Rolling as well, given we did not diverge for Humble yet.

henningkayser, 2022-07-01:
I think we did converge, at least with the sync? Of course we could cut it earlier, though.

Llavrih, 2022-07-13:
Any updates for Humble?

vatanaksoytezer, 2022-07-13:
@Llavrih we will cut a new release for both Humble and Rolling within this week. An eatlier version was already released to Humble and available as binary if you need to use though.

jure-no, 2022-07-15:
Hello, can we expect to get humble version over the weekend?

vatanaksoytezer, 2022-07-15:
@jure-no yes, we will cut the release, but it may take a few days for ros folks to merge our PR to rosdistro, and a few weeks to get it synced to main repositories.

vatanaksoytezer, 2022-08-02:
2.5.2 and 2.5.3 are released, thus closing this.

closed 2022-08-02 by vatanaksoytezer